mavlink/MAVSDK-Python #474: logfile_download.py stuck and unable to download log

kennethakiti1, 2022-04-29:
I don't know if this issue has been resolved yet, but I am facing a similar problem and I can't seem to get any help from the prior discussions.

first, I checked my Qground Control and I can see several log files (22 log files).

now when i run the **[logfile_download.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/logfile_download.py)** in the examples, the execution seems to be stuck for several minutes showing only 

```
Waiting for drone to connect...
-- Connected to drone!

```
this is the case and remains this way until I stop the program execution. 
i can however run other examples like telemetry and the likes.

What could be the problem and how can i view and download the log files

JonasVautherin, 2022-04-29:
Did you try debugging what is happening in the code? For instance, does line 18 ever return? Also did you try checking the output of mavsdk_server when running the script?

kennethakiti1, 2022-04-29:
so i added a few print statements to observe where the code stops

```
#!bin/python3

import asyncio
from mavsdk import System
import sys

async def run():
    drone = System()
    await drone.connect(system_address="serial://COM9:115200")#serial:///dev/ttyACM0:115200

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("waiting for entries ..............")
    entries = await get_entries(drone)
    print(entries)
    for entry in entries:
        print(entry)
        await download_log(drone, entry)

async def download_log(drone, entry):
    date_without_colon = entry.date.replace(":", "-")
    filename = f"./log-{date_without_colon}.ulog"
    print(f"Downloading: log {entry.id} from {entry.date} to {filename}")
    previous_progress = -1
    async for progress in drone.log_files.download_log_file(entry, filename):
        new_progress = round(progress.progress*100)
        if new_progress != previous_progress:
            sys.stdout.write(f"\r{new_progress} %")
            sys.stdout.flush()
            previous_progress = new_progress
    print()

async def get_entries(drone):
    print("inside get_entries")
    entries = await drone.log_files.get_entries()
    print("first entry", entries)
    for entry in entries:
        print(f"Log {entry.id} from {entry.date}")
    return entries

if __name__ == "__main__":
    asyncio.get_event_loop().run_until_complete(run())
```

and the output is

```
pi@raspberrypi:~ $ python3 log_drone.py 
Waiting for drone to connect...
-- Connected to drone!
waiting for entries ..............
inside get_entries
```

showing that it stops execution at 

> entries = await drone.log_files.get_entries()

and checking the output of the mavsdk_server i got 

```
[12:10:34|Info ] MAVSDK version: v1.2.0 (mavsdk_impl.cpp:28)
[12:10:34|Info ] Waiting to discover system on serial:///dev/ttyACM0:115200... (connection_initiator.h:20)
[12:10:34|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:490)
[12:10:34|Debug] Component Autopilot (1) added. (system_impl.cpp:369)
[12:10:35|Debug] Discovered 1 component(s) (system_impl.cpp:570)
[12:10:35|Info ] System discovered (connection_initiator.h:63)
[12:10:35|Debug] MAVLink: info: data link #0 lost (system_impl.cpp:242)
[12:10:35|Info ] Server started (grpc_server.cpp:52)
[12:10:35|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:53)
[12:10:35|Debug] MAVLink: critical: PREFLIGHT FAIL: MAG SENSORS INCONSISTENT (system_impl.cpp:242)
[12:10:35|Debug] MAVLink: info: data link #0 regained (system_impl.cpp:242)
[12:12:08|Warn ] No entries received (log_files_impl.cpp:166)
[12:12:08|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[12:12:09|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[12:12:14|Info ] heartbeats timed out (system_impl.cpp:264)
[12:12:14|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[12:12:16|Debug] Discovered 1 component(s) (system_impl.cpp:570)
[12:12:17|Warn ] sending again after 0.509814 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[12:12:17|Warn ] sending again after 1.01482 s, retries to do: 2  (520). (mavlink_command_sender.cpp:287)
[12:12:18|Warn ] sending again after 1.51821 s, retries to do: 1  (520). (mavlink_command_sender.cpp:287)
[12:12:18|Error] Retrying failed (520) (mavlink_command_sender.cpp:307)
[12:12:19|Info ] heartbeats timed out (system_impl.cpp:264)

```

JonasVautherin, 2022-04-29:
Right, so it seems like MAVSDK is not receiving any entry:

```
[12:12:08|Warn ] No entries received (log_files_impl.cpp:166)
```

You say QGC can list and download log files?

JonasVautherin, 2022-04-29:
Interesting :thinking:. Looks like a bug then :sweat_smile:

julianoes, 2022-05-06:
Thanks for the report. I need to put that on my (growing) todo list :smirk:.

JonasVautherin, 2022-05-09:
Or if someone in the community is willing to have a look at it... :innocent: :see_no_evil:

julianoes, 2023-03-28:
I believe we fixed something recently. I just tried it again and it seems to work now.

closed 2023-03-28 by julianoes

Guruprasath-Codoid, 2025-06-12:
Hey guys,
I’m facing the same issue. I connect my flight controller via serial (USB cable) using Python MAVSDK code on Windows. After a successful connection, when I try to download the log files, I get the following error:

LogFilesError: NO_LOGFILES: 'No Logfiles'; origin: get_entries(); params: ().

However, when I connect the same flight controller in QGroundControl, the log files are listed and I’m able to download them without any problem.

Does anyone have updated or working code for this?