mavlink/MAVSDK-Python #476: Inconsistency in running multiple missions

Turokhunter, 2022-05-20:
Hi, I'm a researcher working on drone data, and I'm trying to make the process of collecting simulation data easier. In that pursuit, I made a python script that executes a series of missions. The architecture that I use is similar to PX4 testing, where the PX4 instance is launched, followed by a mission. However, I noticed that there are times when all 15 missions work fine, and there are times when one of the missions gets hanged upon a connect. I was wondering if there is a way to detect when this happens so that I can either try to reconnect to the vehicle. I'm using PX4, Gazebo, and MAVSDK-Python. 
[image]

julianoes, 2022-05-20:
Which version are you using? I recently found the bug leading to the connection issue: https://github.com/mavlink/MAVSDK/pull/1739

Turokhunter, 2022-05-20:
mavsdk  0.23.0

julianoes, 2022-05-20:
That would explain it. Try with the latest and greatest :).

Turokhunter, 2022-05-20:
I have uninstalled and reinstalled mavsdk 1.3.0.
I run my scripts six times, four times it runs all 15 tests. Once it got hung up on the connecting vehicle and once on starting mission (so at some point during the mission).
[image]
[image]
)
Is there a way to detect these occurrences? So I can stop the run and restart in code? Maybe a timeout feature for connecting to a vehicle. 

I can provide the code if that helps.

julianoes, 2022-05-20:
You can always do workarounds like you said but I would be more interested to fix the actual problems.

The first thing I would do is to run mavsdk_server separately, like described here:
mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

Then you can share the output of when it happens, or you could even run it with `gdb --args path/to/mavsdk_server ...`, and when it stalls, you hit Ctrl+C and in gdb you do: `thread apply all backtrace` and copy the output to here.

Turokhunter, 2022-05-23:
I will try to get you an output in the next two days.

julianoes, 2022-06-03:
> It would be great if MavSDK could detect that everything isn't ready and only start when it is.

This should be improved with https://github.com/mavlink/MAVSDK/pull/1794.

For the connection issue, I would have to see in gdb what is going on as mentioned above:

> Then you can share the output of when it happens, or you could even run it with `gdb --args path/to/mavsdk_server` ..., and when it stalls, you hit Ctrl+C and in gdb you do: `thread apply all backtrace` and copy the output to here.

julianoes, 2022-06-09:
I managed to reproduce something similar where it doesn't connect. I'm investigating. If you could provide details how to reproduce your setup that would still be helpful though.

Turokhunter, 2022-06-09:
I couldn't get the gdb to work. Currently, I'm working on documenting and making my git public. It contains all the code I have been working on. I will post again once I'm done.

Turokhunter, 2022-06-09:
Here is the GitHub with all my setup: https://github.com/VisDunneRight/Px4-Batch-Simulation

julianoes, 2022-06-18:
Thanks I will check it out, hopefully next week.

julianoes, 2022-06-20:
I was able to run this and for me it fails to arm, mostly in the second test:

```
--> Test case 2 of 15: 'Mission 2: Square flight plan' running ...
Connecting to vehicle
Drone discovered!
home location
	>lat:47.3977508
	>lon:8.5456074
-- Uploading mission
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
Traceback (most recent call last):
  File "/home/julianoes/src/upstream/Px4-Batch-Simulation/missions/sdk_square_mission.py", line 114, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/julianoes/src/upstream/Px4-Batch-Simulation/missions/sdk_square_mission.py", line 103, in main
    await mission.arm()
  File "/home/julianoes/src/upstream/Px4-Batch-Simulation/missions/missionHelperSDK.py", line 48, in arm
    await self.vehicle.action.arm()
  File "/home/julianoes/.local/lib/python3.9/site-packages/mavsdk/action.py", line 320, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()
```

```
[10:05:23|Debug] Discovered 1 component(s) (system_impl.cpp:499)
[10:05:23|Warn ] command denied (512). (mavlink_command_sender.cpp:193)
[10:05:24|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:124)
[10:05:24|Debug] MAVLink: info: [logger] ./log/2022-06-20/08_05_21.ulg	 (system_impl.cpp:248)
[10:05:28|Debug] MAVLink: critical: Arming denied! manual control lost (system_impl.cpp:248)
```

Have you seen that too?

julianoes, 2022-06-22:
@Turokhunter you still there?

Turokhunter, 2022-06-22:
Yup, I'm on vacation currently in the UK.

Turokhunter, 2022-06-22:
Nvm. I had a bit of time this morning. Yes, I sometimes get that error and when it occurs it seems to cascade. If I recall you mentioned https://github.com/mavlink/MAVSDK/pull/1794 might improve that from happening. I normally get the issue around mission 4. Mission 1 is a takeoff and land mission. So there are no waypoints set and it doesn't use mission planning.  So mission 2 is the first instance of the mission plan being uploaded and ran. I added a 10-sec timer, for now, to reduce the arm fail but this does not always help. I will do more investigation tonight. I would also suggest using ps. I found quite a bit of residual of mavsdk_server running and even px4_mavlink instances which might be the cause of connecting or arm issue. 

```
--> Test case 12 of 15: 'Mission 12: Square flight plan' running ...
Connecting to vehicle
Drone discovered!
home location
	>lat:47.397750699999996
	>lon:8.5456075
-- Uploading mission
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Starting mission
Traceback (most recent call last):
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_square_mission.py", line 114, in <module>
    loop.run_until_complete(main())
  File "/usr/local/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_square_mission.py", line 106, in main
    await mission.startMission()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/missionHelperSDK.py", line 53, in startMission
    await self.vehicle.mission.start_mission()
  File "/usr/local/lib/python3.9/site-packages/mavsdk/mission.py", line 1099, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()
Process ran too long
killing Px4
Stop_thread set
Rerun test case.
Connecting to vehicle
Drone discovered!
home location
	>lat:47.397750699999996
	>lon:8.5456075
-- Uploading mission
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Starting mission
Traceback (most recent call last):
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_square_mission.py", line 114, in <module>
    loop.run_until_complete(main())
  File "/usr/local/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_square_mission.py", line 106, in main
    await mission.startMission()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/missionHelperSDK.py", line 53, in startMission
    await self.vehicle.mission.start_mission()
  File "/usr/local/lib/python3.9/site-packages/mavsdk/mission.py", line 1099, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()
Process ran too long
killing Px4
Stop_thread set
Rerun test case.
--- Test case 12 of 15: 'Mission 12: Square flight plan' failed.
--> Test case 13 of 15: 'Mission 13: Fly in a circle' running ...
Connecting to vehicle
Drone discovered!
home location
	>lat:47.397750699999996
	>lon:8.5456075
-- Uploading mission
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Starting mission
Traceback (most recent call last):
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_fly_in_circle.py", line 81, in <module>
    loop.run_until_complete(main())
  File "/usr/local/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/sdk_fly_in_circle.py", line 73, in main
    await mission.startMission()
  File "/Users/turokhunter/Dropbox/Work/Drone-Simulation/missions/missionHelperSDK.py", line 53, in startMission
    await self.vehicle.mission.start_mission()
  File "/usr/local/lib/python3.9/site-packages/mavsdk/mission.py", line 1099, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()
```

julianoes, 2022-06-22:
> I found quite a bit of residual of mavsdk_server running and even px4_mavlink instances which might be the cause of connecting or arm issue.

But wouldn't that immediately lead to a `bind` error?

Turokhunter, 2022-06-23:
Here is an instance of what I was describing with the ps.  The mavsdk_server occurs when the connection fails. I think for each failure a new mavsdk_server is created until some level of capacity is reached.  The other possibility is once you have a failed connection, all missions afterward fail to connect.

```
--- Test case 9 of 15: 'Mission 9: Fly in random direction 4 times' succeeded.
--> Test case 10 of 15: 'Mission 10: Fly in a square search pattern' running ...
Connecting to vehicle
Process ran too long
killing Px4
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 10 of 15: 'Mission 10: Fly in a square search pattern' failed.
--> Test case 11 of 15: 'Mission 11: Takeoff and land' running ...
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 11 of 15: 'Mission 11: Takeoff and land' failed.
--> Test case 12 of 15: 'Mission 12: Square flight plan' running ...
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 12 of 15: 'Mission 12: Square flight plan' failed.
--> Test case 13 of 15: 'Mission 13: Fly in a circle' running ...
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 13 of 15: 'Mission 13: Fly in a circle' failed.
--> Test case 14 of 15: 'Mission 14: Fly in random direction 4 times' running ...
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 14 of 15: 'Mission 14: Fly in random direction 4 times' failed.
--> Test case 15 of 15: 'Mission 15: Fly in a square search pattern' running ...
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
Connecting to vehicle
Process ran too long
Stop_thread set
Rerun test case.
--- Test case 15 of 15: 'Mission 15: Fly in a square search pattern' failed.
Tariks-MBP:Drone-Simulation turokhunter$ ps
  PID TTY           TIME CMD
 1346 ttys000    0:00.06 -bash
 1396 ttys001    0:00.06 -bash
 1723 ttys002    0:00.07 -bash
 1767 ttys003    0:00.08 -bash
 1846 ttys004    0:00.07 -bash
 1990 ttys005    0:00.06 -bash
 2036 ttys006    0:00.07 -bash
  194 ttys007    0:08.70 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
  521 ttys007    0:08.22 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
  752 ttys007    0:07.69 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
  959 ttys007    0:07.28 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 1106 ttys007    0:06.87 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 1263 ttys007    0:06.39 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 1500 ttys007    0:05.96 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 1591 ttys007    0:05.50 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 2375 ttys007    0:00.29 -bash
99574 ttys007    0:00.02 cmake --build /Users/turokhunter/Dropbox/Work/Drone-Simulation/px4Developer/Firmware/build/px4_sit
99575 ttys007    0:00.45 /usr/local/bin/ninja gazebo
99586 ttys007    0:00.00 /bin/sh -c cd /Users/turokhunter/Dropbox/Work/Drone-Simulation/px4Developer/Firmware/build/px4_sit
99587 ttys007    0:00.01 bash /Users/turokhunter/Dropbox/Work/Drone-Simulation/px4Developer/Firmware/Tools/sitl_run.sh /Use
99606 ttys007  225:11.25 /Users/turokhunter/Dropbox/Work/Drone-Simulation/px4Developer/Firmware/build/px4_sitl_default/bin/
99607 ttys007    0:00.05 /bin/sh etc/init.d-posix/rcS 0
99690 ttys007    0:00.00 px4-mavlink --instance 0 stream -r 50 -s SERVO_OUTPUT_RAW_0 -u 18570
99692 ttys007    0:09.45 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
99776 ttys007    0:09.29 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
99863 ttys007    0:09.06 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
99951 ttys007    0:08.86 /usr/local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server -p 50051 --sysid 245 --compid 190 
 2716 ttys008    0:00.05 -bash
Tariks-MBP:Drone-Simulation turokhunter$ 
```
As to the question about the bind I'm not sure.  You can't run the script until running `killall px4-mavlink`. I noticed that if the mavsdk_server is still running then you get the `mavsdk.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()`, 
Running the `killall mavsdk_server` resets back to default state until running into another connect fail.