mavlink/MAVSDK-Python #478: Virtual Joystick for multiple drones

bernardileonardo, 2022-05-31:
Hi, to switch a drone in OFFBOARD mode using a simulator (Gazebo) I need to have a virtual joystick enabled for it in QGroundControl. Now I'm trying to control multiple drones with same code using OFFBOARD mode but only the first drone switch in such mode because QGroundControl provides virtual joystick only for first drone. How can i switch other drones in OFFBOARD mode?

JonasVautherin, 2022-05-31:
Are you using MAVSDK or is this a QGC question?

bernardileonardo, 2022-05-31:
So I can ask, How can I use OFFBOARD mode using a Simulator?

julianoes, 2022-06-01:
Can you try to take a step back and describe the problem/challenge you are trying to solve. That way we have a bit more context and are likely better able to come up with a good suggestion.

bernardileonardo, 2022-06-01:
> Can you try to take a step back and describe the problem/challenge you are trying to solve. That way we have a bit more context and are likely better able to come up with a good suggestion.

I tried offboard examples and they didn't work, drone doesn't switch in offboard mode because of "failsafe enabled: No manual control stick input". I'm in a simulated environment, i solved the problem for one drone checking groundcontrol box for "virtual joystick". Now I'm trying to control multiple drones, how can overcome the fact that I can not have multiple virtual joystick ? I have the previous problem except for the first drone for Wich i have a virtual joystick.

julianoes, 2022-06-01:
What are you trying to do? Fly offboard, fly manual? Fly with 3 drones in synchrony? Have one drone lead, and 3 drones follow? How are they connected? On the same network? Do they have different MAVLink IDs? Is it automated flight using MAVSDK? Or are you trying to fly with 3 drones in QGC? Fixedwing, quadcopter?
This is what I mean with context.

For the simple issue where you need manual input for offboard, but you don't have it, you might just want to use the param:
https://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_RCL_EXCEPT

bernardileonardo, 2022-06-01:
I'm trying to fly offboard. At first i tried with one drone, then with 3 of them

JonasVautherin, 2022-06-01:
Did you check the param suggested by Julian? Does it help?

bernardileonardo, 2022-06-01:
Thanks, param works for drone 2 and 3. The first drone still generates the same error(even with the param set to offboard) but I can use a virtual joystick for it

julianoes, 2022-06-02:
I believe it will work for drone 1 as well as long as you never connect RC in the first place and make sure the param is correct. I can't think of a reason why it would not work for drone 1.

I take it your problem is solved then and I can close this issue.

closed 2022-06-02 by julianoes