mavlink/MAVSDK-Python #479: Drones Communication

bernardileonardo, 2022-06-08:
Hi, is there the possibility making drones communicates with each other? For example exchanging position

bernardileonardo, 2022-06-09:
> That's certainly possible. Each MAVLink message contains a system ID and component ID of the source, so if two drones are in the same communication channel / network, they can send position to each other using MAVLink, as long as they don't have the same system ID.

So i need to use mavlink or are there any mavsdk-python function that permit sending info to another drone specifing it's Id?

julianoes, 2022-06-18:
I would need some more context and explanation of your architecture to be able to help you.

bernardileonardo, 2022-06-20:
I have many drones to control over mavsdk-python, i need them communicating each others information like their position or other custom messages but also position or bool value is enough. I'm not able to find any function to send mavlink messages using mavsdk-python. Are there any ones? I don't know how to be more explicative

julianoes, 2022-06-20:
Ok I understand you want to send information like position from one drone to another one. So this means you either need to forward such messages manually to other drones or make use of network broadcasting where everyone can receive anyone else's messages.

Then my question would be what a drone does with that information, or rather what you have implemented to receive such information.

And to answer your question: no mavsdk does not support that sort of thing, as it has not been a use case. However, if you want one drone to listen to another one, you could - in theory - use mavsdk on the other drone acting like a ground station and receiving messages, e.g. telemetry. Then you still have to actually do something given that information.

bernardileonardo, 2022-06-22:
I'm using drones to search for a target over an area and communicate to other drones if one of them found it to stop other drones searching. At first I was thinking to implement some sort of floacking but for the moment I am controlling them from the companion computer independently of each other. Once the computer knows that a drone has identified the target it stopped searching for the other drones

julianoes, 2022-06-22:
Right, so then the question is whether you have a central unit controlling them all or whether it's distributed and they all process each others information.

Jai-GAY, 2024-02-20:
@bernardileonardo  , I think the soon-to-be-released firmware may be of interest if you use the [Ardupilot stack](https://discuss.ardupilot.org/t/ethernet-and-tcp-ip-in-ardupilot/109779).