mavlink/MAVSDK-Python #481: 'Unsupported'; origin: set_pitch_rate_and_yaw_rate(); params: (10, 0)" IN VERSION 1.3.0

MalikMuneebAhmad, 2022-06-21:
I am running make px4_sitl gazebo_typhoon_h480. To control the gimbal, I am running gimbal.py from MAVSDK-Python but following error occured "raise GimbalError(result, "set_pitch_rate_and_yaw_rate()", pitch_rate_deg_s, yaw_rate_deg_s) mavsdk.gimbal.GimbalError: UNSUPPORTED: 'Unsupported'; origin: set_pitch_rate_and_yaw_rate(); params: (10, 0)"

julianoes, 2022-06-21:
What's the PX4 version?

MalikMuneebAhmad, 2022-06-22:
I have installed it from source.

MalikMuneebAhmad, 2022-06-22:
Details:
M	Tools/sitl_gazebo
M	launch/posix_sitl.launch
Your branch is up to date with 'origin/master'.

julianoes, 2022-06-22:
I see, master, ok. I'll try to reproduce it.

julianoes, 2022-06-22:
And what's the last / top commit when you do `git log`?

julianoes, 2022-06-22:
@MalikMuneebAhmad I tested this and for me it just works.

Can you also type `gimbal status` into the `pxh>` to check the output please?

For me, after running the script, it is:
```
pxh> gimbal status
INFO  [gimbal] Input Selected
INFO  [gimbal] Input: Mavlink (Gimbal V2)
INFO  [gimbal] Input not selected
INFO  [gimbal] Input: Test
  roll :   0 deg
  pitch:   0 deg
  yaw  :   0 deg
INFO  [gimbal] Output: MAVLink gimbal protocol v2
  quaternion: [1.0 0.0 0.0 0.0]
  angular velocity: [nan nan nan]

```

MalikMuneebAhmad, 2022-06-23:
Output of git log is 

Author: Daniel Agar <daniel@agar.ca>
Date:   Sat Jun 4 15:58:56 2022 -0400

    Update submodule GPSDrivers to latest Sat Jun  4 12:38:49 UTC 2022 (#19766)
    
        - GPSDrivers in PX4/Firmware (6e77a084cd43830c8b13018b7fd5470da9bc4ff5): https://github.com/PX4/PX4-GPSDrivers/commit/181fae1a4b5e33576d786755782adb2f195ecc48
        - GPSDrivers current upstream: https://github.com/PX4/PX4-GPSDrivers/commit/016c37cd1f18c716427e2465d8daa6aa1054b0f1
        - Changes: https://github.com/PX4/PX4-GPSDrivers/compare/181fae1a4b5e33576d786755782adb2f195ecc48...016c37cd1f18c716427e2465d8daa6aa1054b0f1
    
        016c37c 2022-06-01 Julian Oes - sbf: fix overrun on i

MalikMuneebAhmad, 2022-06-23:
gimbal status is

pxh> gimbal status
INFO  [gimbal] Input Selected
INFO  [gimbal] Input not selected
INFO  [gimbal] Input: Test
  roll :   0 deg
  pitch:   0 deg
  yaw  :   0 deg
INFO  [gimbal] Input: Mavlink (CMD_MOUNT)
INFO  [gimbal] Output: MAVLink gimbal protocol v1

MalikMuneebAhmad, 2022-06-23:
After running `gimbal test yaw 30`  gimbal status is changing but drone gimbal is still not changing.

INFO  [gimbal] Input Selected
INFO  [gimbal] Input not selected
INFO  [gimbal] Input: Test
  roll :   0 deg
  pitch:   0 deg
  yaw  :  30 deg

julianoes, 2022-06-23:
Ok, so the first step is to get the gimbal working with the test command. What sort of gimbal is it and how is it connected?

You can use the param [MNT_MODE_OUT](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MNT_MODE_OUT) to set the protocol between autopilot and gimbal.

MalikMuneebAhmad, 2022-06-23:
Thank for your continuous replies. I have set the MNT_MODE_OUT and its value is 
Symbols: x = used, + = saved, * = unsaved
x   MNT_MODE_OUT [431,941] : 2
but  `gimbal status` is still showing v1
INFO  [gimbal] Input: Mavlink (CMD_MOUNT)
INFO  [gimbal] Output: MAVLink gimbal protocol v1

julianoes, 2022-06-24:
Nice, so first step done. Now what about trying the gimbal.py example again?

MalikMuneebAhmad, 2022-06-25:
I have commented the lines `drone.gimbal.set_pitch_rate_and_yaw_rate(10, 0)` in gimble.py and it is working and gimble is operating accordingly. But how can we change the pitch and yaw rate.?

julianoes, 2022-06-30:
So you want to set pitch angle and yaw rate?

What about first setting pitch and yaw: and then just set yaw rate (and leave pitch rate at 0) ?