mavlink/MAVSDK-Python #482: Read data from multiple distance sensor

bernardileonardo, 2022-06-22:
Hi, I have a gazebo model that includes multiple laser sensors:

<sensor name="multiplelaser" type="ray">
        <pose>0 0 0 0 1.57079633 0</pose>
        <ray>
          <scan>
            <horizontal>
              <samples>6</samples>
              <resolution>1</resolution>
              <min_angle>-0.3</min_angle>
              <max_angle>0.3</max_angle>
            </horizontal>
          </scan>
          <range>
            <min>0.06</min> 
            <max>35</max>   
            <resolution>0.01</resolution>
          </range>
        </ray>
        <plugin name="LaserPlugin" filename="libgazebo_lidar_plugin.so">
          <robotNamespace></robotNamespace>
          <min_distance>0.2</min_distance>
          <max_distance>15.0</max_distance>
        </plugin>
        <always_on>1</always_on>
        <update_rate>20</update_rate>
        <visualize>true</visualize>
      </sensor>

If I use the MAVSDK-Pyton function mavsdk.telemetry.distance_sensor() I obtain value only from the first ray, how can i read data from the others?

julianoes, 2022-06-22:
That's currently not supported. A pull request would be welcome to improve this.

The first step would be to adapt the API and change this line here:
https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/telemetry/telemetry.proto#L283

To a `repeated DistanceSensor distance_sensors = 1` so that it can contain several.

Murali-IsPagro, 2022-08-13:
> Hi, I have a gazebo model that includes multiple laser sensors : samples tag of tag scan of tag ray:6; plugin=libgazebo_lidar_plugin.so
> 
> If I use the MAVSDK-Pyton function mavsdk.telemetry.distance_sensor() I obtain value only from the first ray, how can i read data from the others?

@bernardileonardo
I had a same scenario I have used two sensor (LIDAR,) . I will explain below how I got multiple sensor data. It might useful for someone.

1. TFMini connected to GPS2 port in Pixhawk cube (configured as SENS_TFMINI_CFG -> GPS2)
2. Terarange One connected to I2C2 .(SENS_EN_TRANGER)

Next, I checked i2c is detected or not using "i2cdetect -b 2" It detects the address
Then, "ls /obj" found that distance_sensor 0 and distance_sensor 1 are publishing.
Last, "listener distance_sensor" 

[image]

In MAVSDK-Python, I used  telemetry.distance_sensor():

>
    async for distance in drone.telemetry.distance_sensor():
        for i in range(2):
           print(f"distance %d: {distance.current_distance_m}"%i)
[image]

I got as two separate distance_sensor data.

[image]