mavlink/MAVSDK-Python #484: RTK client msg FROM ntrip server

g1tm4n123, 2022-06-22:
can you show me an example, the use of mavsdk RTK plugin?

i connected to the vehicle on serial port and i have an ntrip server
await PXvhc.connect(system_address="serial:///dev/ttyUSB0:...")
ntrip caster xxx.xxx.xxx.xx
ntrip port 2101
ntrip mountpoint blox
ntrip username user
ntrip password pwd

Vehicle: Pixhawk
RTK rover: F9P

_setup_stub(self, channel)
channel?
send_rtcm_data(self, rtcm_data)
from ntrim to rtcm_data?

JonasVautherin, 2022-06-22:
@CY-1992 @Katawann @AlexisTM: Would one of you be willing to contribute an example? :blush:

Katawann, 2022-06-22:
Hi @g1tm4n123,

I used the RTK plugin with MAVSDK-C++, but you should be able to make it work the same way with Python. If you can get the RTCM data, you need to send the information with `send_rtcm_data` back to your FCU. You also have to send your _GPGGA_ data back to the Ntrip server you are connected to every X seconds

The Ntrip client from MAVProxy [here](https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/lib/ntrip.py) could be helpful to make sure you did not forget anything

Also there was a bug in the RTK plugin  [here](https://github.com/mavlink/MAVSDK/issues/1800) but which has been fixed since but not sure if MAVSDK-Python already have this fix. It means you need to truncated the data by yourself before to send it with the MAVSDK plugin. A good example can be found in the MAVProxy code [here](https://github.com/ArduPilot/MAVProxy/blob/7258994299be1e9a62f30a3f5b2498e9fb7a69bd/MAVProxy/modules/mavproxy_ntrip.py#L103)

Let me know if it works !

g1tm4n123, 2022-06-22:
I can't run MAVProxy and MAVSDK together
when MAVDSK running: MAVProxy says: NO LINK
when MAVPROXY running: MAVSDK says: not connect
 
I want to use only MAVSDK

PX Vehicle => Linux USB port
F9P RTK rover => PX Vehicle TELEM2 port
ntrip running: UBLOX server

According to [these simple example](https://github.com/mavlink/MAVSDK/tree/main/examples),  I was able to solve many tasks.
Can you show something like this in these ntrip MAVSDK topics?

AlexisTM, 2022-06-23:
I don't have a setup to test it 😞

julianoes, 2022-06-23:
@JonasVautherin v1.4.2 doesn't include the [RTK fix](https://github.com/mavlink/MAVSDK/pull/1808) yet. I will make another backported release.

julianoes, 2022-06-23:
https://github.com/mavlink/MAVSDK/releases/tag/v1.4.3

julianoes, 2023-06-06:
The MAVSDK-Python Rtk plugin should now work.

This is the [fix on the C++ side](https://github.com/mavlink/MAVSDK/pull/2073).

You need mavsdk_server version [v1.4.16](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.16) and MAVSDK-Python [v1.4.8](https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.8).

Example how to use the Rtk Plugin in Python: [rtk.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/rtk.py).

closed 2023-06-06 by julianoes