mavlink/MAVSDK-Python #486: uable to use MAV_CMD_DO_SET_SERVO

KBhanup, 2022-06-23:
I am trying to use MAV_CMD_DO_SET_SERVO to send PWM commands to AUX1 (channel 9). But unable to find the right protocol structure to do it.
Aim: I want to integrate activating and Deactivating magnet into a FCU code that I have built (set Magnet at AUX1 normally operated through RC)

 msg_hi = self.vehicle.message_factory.command_long_encode(
                0, 0,   # target_system, target_command
                mavutil.mavlink.MAV_CMD_DO_SET_SERVO, # command
                0,
                8,    # servo number
                2006, # servo position
                0, 0, 0, 0, 0)

Here, I know here in this example I have tried using similar dronekit function not knowing how I can do using MAVSDK or MAVROS,
Pleaseee guide me if any alternative ways how I can do it?

%%
#!/usr/bin/python

import rospy as rp
import threading

from dronekit import connect, VehicleMode
from pymavlink import mavutil

from std_msgs.msg import Int8
from sensor_msgs.msg import Joy
from geometry_msgs.msg import PoseStamped, TwistStamped

class magdroneControlNode():

    def __init__(self):
        rp.init_node("magdrone_deploy")

        # Connect to the Vehicle
        rp.loginfo('Connecting to Vehicle')
        self.vehicle = connect('/dev/ttyUSB0', wait_ready=True, baud=921600)

        # Set up Subscribers
        self.joy_sub = rp.Subscriber("/joy", Joy, self.joy_callback, queue_size=1)

        # Create thread for publishers
        self.rate = 30
        t = threading.Thread(target=self.send_commands)
        t.start()

        rp.spin()

    def joy_callback(self, data):
        # Button Controls
        if data.buttons[0] == 1.0:
            rp.loginfo("Mission set to 1 - Deploying Sensor")
            self.mission_id = 1
            
        if data.buttons[1] == 1.0:
            rp.loginfo("Mission set to 2 - Retrieving Sensor and Exiting")
            self.mission_id = 2
            
    '''
        Magnet Control
    '''

    def engage_magnet(self):
        msg_hi = self.vehicle.message_factory.command_long_encode(
                0, 0,   # target_system, target_command
                mavutil.mavlink.MAV_CMD_DO_SET_SERVO, # command
                0,
                8,    # servo number
                2006, # servo position
                0, 0, 0, 0, 0)

        msg_neut = self.vehicle.message_factory.command_long_encode(
                0, 0,   # target_system, target_command
                mavutil.mavlink.MAV_CMD_DO_SET_SERVO, # command
                0,
                8,    # servo number
                1500, # servo position
                0, 0, 0, 0, 0)

        # Send command
        self.vehicle.send_mavlink(msg_hi)
        rp.loginfo("Magnet Engaged")
        time.sleep(5)
        self.vehicle.send_mavlink(msg_neut)
        rp.loginfo("Magnet in Neutral")
        self.docked = True

    def disengage_magnet(self):
        msg_low = self.vehicle.message_factory.command_long_encode(
                0, 0,   # target_system, target_command
                mavutil.mavlink.MAV_CMD_DO_SET_SERVO, # command
                0,
                8,    # servo number
                982,  # servo position
                0, 0, 0, 0, 0)

        msg_neut = self.vehicle.message_factory.command_long_encode(
                0, 0,   # target_system, target_command
                mavutil.mavlink.MAV_CMD_DO_SET_SERVO, # command
                0,
                8,    # servo number
                1500, # servo position
                0, 0, 0, 0, 0)

        # Send command
        self.vehicle.send_mavlink(msg_low)
        rp.loginfo("Magnet Disengaged")
        time.sleep(5)
        self.vehicle.send_mavlink(msg_neut)
        rp.loginfo("Magnet in Neutral")
        self.docked = False

       

    '''
        Main Loop
    '''

    def send_commands(self):
        rp.loginfo("Accepting Commands")
        r = rp.Rate(self.rate)
        while not rp.is_shutdown():   

            if (self.mission_id == 1) & (not self.magnet_engaged):
                self.magnet_engaged = True
                t = threading.Thread(target=self.engage_magnet)
                t.start()

            if (self.mission_id == 2) & (self.magnet_engaged):
                self.magnet_engaged = False
                t = threading.Thread(target=self.disengage_magnet)
                t.start()
            r.sleep()

# Start Node
magdrone = magdroneControlNode()

JonasVautherin, 2022-06-23:
I am not familiar with `MAV_CMD_DO_SET_SERVO`. Is it similar to MAVSDK's [SetActuator](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L101-L104) action?

julianoes, 2022-06-23:
It looks like this is a PX4 question and not a MAVSDK question. Also your example code is DroneKit or pymavlink and not MAVSDK, so not sure what to do with it.

closed 2022-06-23 by julianoes