mavlink/MAVSDK-Python #488: RTCM packages

xinadax2000, 2022-07-10:
I have RTCM messages in Binary Array (from mavproxy)
How can I translate it into mavsdk RtcmData?
https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/rtk.py
I want to send a message to the orangecube
send_rtcm_data(rtcm) => binary_array error

julianoes, 2022-07-11:
I don't know what sort of binary mavproxy sends. I'd assume it's just the same RTCM.

What is this binary_array error? Can you paste the exact error please? I assume it's some sort of Python type error.

xinadax2000, 2022-07-16:
code:
await drone.rtk.send_rtcm_data(rtcm_data)

error:
AttributeError: 'bytearray' object has no attribute 'translate_to_rpc'

rtcm_data:
bytearray(b'\xd3\x00mCP\x00\x8c2\x16\x82\x00\x00,@\x88\x00\x00\x00\x00\x00 \x00\x00\x00~\x9c\xa4\x9a\x90\xa2\x8c\x00\x00\x01\xa7\xa2\x1e=gv\x8f\x1fq{\\\x13_\xc9\xdf\x17\x02L$\xb6\xdd\x17\x9a.\xe8\xba\x94\x02U6^\xa2^\x08\xac\xf5\xf4\x1d\xcc\n\x9d\xe7\xeb\x04R\x15\x92\x93\xf9o\xf2\xc1\xb5-j\xba\xf12`@\r\x83\xc0\xe8B\x0f\x05\xec\x8c\xfc\xc4\x88l\xac\x7f\xf1\x1aR\xc2\xbc\x87')

julianoes, 2022-07-18:
Can you share the sample code how you are using the API?

xinadax2000, 2022-07-18:
I use this ntrip client: https://github.com/jcmb/NTRIP/blob/master/NTRIP%20Client/NtripClient.py

julianoes, 2022-07-21:
Duplicate of #491, let's continue there.

closed 2022-07-21 by julianoes