mavlink/MAVSDK-Python #490: grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with: 	status = StatusCode.UNAVAILABLE 	details = "Socket closed" 	debug_error_string = "{"created":"@1657699376.171754659","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1055,"grpc_message":"Socket closed","grpc_status":14}"

Berruaa, 2022-07-13:
Hello! I am using Rasperry Pi 4B (Raspbian GNU/Linux 11) connecting a drone with Pixhawk 4 (PX4 v1.13.0 firmware), using FTDI FT232BL TTL to USB.
The drone can be armed but I keep getting this error message when I run the script:
`# Python 3.9.2

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/ttyPixhawk:921600")
    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("-- Arming")
    await drone.action.arm()
    await asyncio.sleep(3) 

    print("-- Disarming")
    await drone.action.disarm()
    await asyncio.sleep(2)
    
    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())`
[image]

When I run the mavsdk_server:

[image]

Could you help me with issue. Thank you very much!

JonasVautherin, 2022-07-13:
Hmm something seems weird with the connection to `mavsdk_server`. Can you try running `mavsdk_server` manually (`./mavsdk_server serial:///dev/ttyPixhawk:921600`), and connect to it with:

```python
drone = System(mavsdk_server_address="localhost")
await drone.connect()
```

Berruaa, 2022-07-14:
It works! Thank you very much!

closed 2022-07-14 by Berruaa

JonasVautherin, 2022-07-14:
Did you run the embedded `mavsdk_server`, or did you download it?

I'm just wondering why your first version has this issue :thinking:. Which version of MAVSDK are you using?