mavlink/MAVSDK-Python #491: RtcmData

xinadax2000, 2022-07-18:
How can I make a  MAVsdk RtcmData out of RTCM ByteArray?
I get the RTCM ByteArray from u-center ntrip.

julianoes, 2022-07-18:
What format are you getting out of u-center ntrip?

xinadax2000, 2022-07-18:
I use this ntrip client: https://github.com/jcmb/NTRIP/blob/master/NTRIP%20Client/NtripClient.py

def readLoop(self):
		while True:
			data = self.read()
			if data is None:
				continue
			asyncio.run(self.send_data(data))

#data = self.rtcm3.get_packet()

async def send_data(self, data):
		await self.drone.rtk.send_rtcm_data(data)

julianoes, 2022-07-21:
Ok, and what's the type of the data from `self.read()`?

julianoes, 2022-07-21:
From other issue:

```
code:
await drone.rtk.send_rtcm_data(rtcm_data)

error:
AttributeError: 'bytearray' object has no attribute 'translate_to_rpc'

rtcm_data:
bytearray(b'\xd3\x00mCP\x00\x8c2\x16\x82\x00\x00,@\x88\x00\x00\x00\x00\x00 \x00\x00\x00~\x9c\xa4\x9a\x90\xa2\x8c\x00\x00\x01\xa7\xa2\x1e=gv\x8f\x1fq{\\x13_\xc9\xdf\x17\x02L$\xb6\xdd\x17\x9a.\xe8\xba\x94\x02U6^\xa2^\x08\xac\xf5\xf4\x1d\xcc\n\x9d\xe7\xeb\x04R\x15\x92\x93\xf9o\xf2\xc1\xb5-j\xba\xf12`@\r\x83\xc0\xe8B\x0f\x05\xec\x8c\xfc\xc4\x88l\xac\x7f\xf1\x1aR\xc2\xbc\x87')
```

julianoes, 2022-07-21:
So you're presumably passing a `bytearray` to `send_rtcm_data()`.

However, `send_rtcm_data` wants a `RtcmData` type.

And `RtcmData` wants to be initialized with a string.

See: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/rtk.html#mavsdk.rtk.RtcmData

I then tried to line up the types:
```
#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk.rtk import RtcmData

async def send_data(data):
    drone = System()
    await drone.connect()
    await drone.rtk.send_rtcm_data(data)

if __name__ == '__main__':
    rtcm_data = bytearray(b'\xd3\x00mCP\x00\x8c2\x16\x82\x00\x00,@\x88\x00\x00\x00\x00\x00 \x00\x00\x00~\x9c\xa4\x9a\x90\xa2\x8c\x00\x00\x01\xa7\xa2\x1e=gv\x8f\x1fq{\\x13_\xc9\xdf\x17\x02L$\xb6\xdd\x17\x9a.\xe8\xba\x94\x02U6^\xa2^\x08\xac\xf5\xf4\x1d\xcc\n\x9d\xe7\xeb\x04R\x15\x92\x93\xf9o\xf2\xc1\xb5-j\xba\xf12`@\r\x83\xc0\xe8B\x0f\x05\xec\x8c\xfc\xc4\x88l\xac\x7f\xf1\x1aR\xc2\xbc\x87')
    asyncio.run(send_data(RtcmData(str(rtcm_data))))
```

And that worked, except that the mavsdk_server v1.4.3 which comes with the pip install doesn't support the Rtk plugin yet, so I will fix that.

```
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
```

julianoes, 2022-07-21:
Fix coming here: https://github.com/mavlink/MAVSDK/pull/1865

I'll release it to mavsdk v1.4.3 and to the Python wrappers as well ASAP.

julianoes, 2022-07-21:
mavsdk on PyPy should now be updated to version 1.4.1 containing mavsdk_server version v1.4.4 including the Rtk plugin.

julianoes, 2022-07-22:
I'm closing this but just comment again if it's not resolved for you.

closed 2022-07-22 by julianoes

swimmingseeds, 2023-02-21:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/gps-rtcm-data-not-recognized-from-f9p-chip-u-center/30915/1