mavlink/MAVSDK-Python #492: Mission starts when upload_mission called intead of start_mission

m-sienk, 2022-07-21:
I upload the first mission, arm and then start the mission. This performs as expected.
While still hovering in place I upload a second mission, the second mission begins immediately after the upload has completed without me calling 'start mission'.

Is this intended functionality? if so its somewhat confusing 

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk.mission import (MissionItem, MissionPlan)

def first_mission():
    mission_items = []
    mission_items.append(MissionItem(47.39786113590441,
                                     8.545539403670114,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.39779940558625,
                                     8.545492465012353,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.39771588798181,
                                     8.545448208563608,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))

    return MissionPlan(mission_items)

def second_mission():
    mission_items = []
    mission_items.append(MissionItem(47.39781075307662,
                                     8.545821035616678,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.39772995889185,
                                     8.545770073645395,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.39766096600676,
                                     8.545716429465097,
                                     3,
                                     5,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan')))

    return MissionPlan(mission_items)

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("-- Uploading first mission")
    await drone.mission.upload_mission(first_mission())

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Starting mission")
    await drone.mission.start_mission()

    print("Waiting until first mission finished...")
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: "
              f"{mission_progress.current}/"
              f"{mission_progress.total}")

        if mission_progress.current == mission_progress.total:
            break

    print("-- Uploading second mission")
    await drone.mission.upload_mission(second_mission())

    print("Waiting until second mission finished...")
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: "
              f"{mission_progress.current}/"
              f"{mission_progress.total}")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

julianoes, 2022-07-21:
Is this with ArduPilot or PX4?

With PX4 it will start the mission if it still is in mission mode. I'm not sure from the top of my head if it stays in mission mode at the end of a mission, or if it switches to Hold/Loiter mode.

m-sienk, 2022-07-21:
I'm using PX4. It remains in mission mode after a mission has completed.

Thanks for taking the time to explain, I fixed this by using your suggestion and calling hold after the first mission has completed.
Doesn't it make more sense for it swap back to Hold/Loiter after the mission has completed?

closed 2022-07-22 by m-sienk

julianoes, 2022-07-26:
I'd say it would make sense, yes. I'm putting that on my list to look into.