mavlink/MAVSDK-Python #493: receiving telemetry during a mission

fadhlimac, 2022-07-21:
hello , I have a question regarding telemetry
>>> import asyncio
>>> from mavsdk import System
>>> drone = System()
>>> await drone.connect()
>>> await drone.action.arm()
Then I took off from QGC to start the mission, How am I supposed to receive the live telemetry position using mavsdk?
because I wrote 
>>>async for position in drone.telemetry.position()
and at the command line it wrote requires indentation, is there something aI did wrong, really appreciate if someone could clarify

JonasVautherin, 2022-07-22:
Did you try the examples? Some of them do exactly that.

julianoes, 2022-07-26:
What exactly are you trying to do? Maybe you can give some background. Also, when pasting code here, make sure to put it within ``` to have it formatted properly.