mavlink/MAVSDK-Python #495: RTK PX4

xindax2000, 2022-07-28:
THX for your help so far!
I send the RTCM messages from the ntrip caster by [MASDK ](https://github.com/mavlink/MAVSDK-Python/issues/491) to the PX4. The messaging works also perfectly. But the GNSS still don't turn into RTK. It works as a normal GPS. (Sirius RTK rover) What else needs to be set up?

julianoes, 2022-07-29:
I don't know, honestly. I can only suggest to ask in PX4 Slack if someone else is using this, or to try to debug the status.

I assume there is some sort of config that you need to do on the PX4 side?

Or maybe the data as we send it is garbage/not the right format.

dayjaby, 2022-07-29:
Which RTCM message IDs do you receive/send forward to the GPS? From what I know with ublox chips you need at least 1005, 1077, 1087 *and* 1230 for getting RTK float/fixed. If you are missing one of these, you only get differential GPS.

xindax2000, 2022-07-29:
It sends them all.
If I send the RTCM messages to the device in the u-center without PX4, it works perfectly in RTK. On the PX4 website it is written that it should go automatically, but there they write two receivers.

xindax2000, 2022-07-30:
Maybe it really didn't send everything. With another solution, I get a large RTCM message. 
But with this solution I get an error like this:
Exception has occurred: RtkError
TOO_LONG: '**Too Long**'; origin: send_rtcm_data(); params: (<mavsdk.rtk.RtcmData object at 0x7f6c7ff89f10>,)
RTCM DATA: array('B', [211, 0, 195, 67, 80, 0, 139, 71, 123, 38, 0, 0, 41, 64, 136, 2, 0, 0, 0, 0, 32, 0, 128, 0, 91, 210, 156, 162, 136, 142, 162, 140, 142, 0, 0, 0, 20, 148, 195, 103, 118, 35, 211, 231, 84, 126, 161, 237, 144, 4, 127, 52, 252, 180, 10, 104, 52, 188, 98, 14, 56, 249, 67, 131, 126, 5, 238, 216, 70, 95, 130, 173, 62, 54, 203, 251, 173, 167, 40, 165, 236, 36, 132, 12, 172, 94, 12, 149, 44, 4, 153, 149, 224, 219, 177, 224, 16, 161, 249, 42, 218, 0, 109, 230, 27, 143, 58, 212, 69, 145, 99, 217, 22, 69, 145, 18, 210, 160, 0, 37, 11, 194, 208, 128, 50, 11, 130, 144, 112, 26, 12, 156, 28, 188, 185, 78, 28, 9, 240, 47, 223, 91, 222, 192, 101, 148, 141, 96, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8, 127, 220])

dayjaby, 2022-07-30:
The array looks good. 211 is the preamble. 0 can be ignored. 195 is the data length. Then you have 195 data bytes and 3 bytes CRC.

```
>>> a = [211, 0, 195, 67, 80, 0, 139, 71, 123, 38, 0, 0, 41, 64, 136, 2, 0, 0, 0, 0, 32, 0, 128, 0, 91, 210, 156, 162, 136, 142, 162, 140, 142, 0, 0, 0, 20, 148, 195, 103, 118, 35, 211, 231, 84, 126, 161, 237, 144, 4, 127, 52, 252, 180, 10, 104, 52, 188, 98, 14, 56, 249, 67, 131, 126, 5, 238, 216, 70, 95, 130, 173, 62, 54, 203, 251, 173, 167, 40, 165, 236, 36, 132, 12, 172, 94, 12, 149, 44, 4, 153, 149, 224, 219, 177, 224, 16, 161, 249, 42, 218, 0, 109, 230, 27, 143, 58, 212, 69, 145, 99, 217, 22, 69, 145, 18, 210, 160, 0, 37, 11, 194, 208, 128, 50, 11, 130, 144, 112, 26, 12, 156, 28, 188, 185, 78, 28, 9, 240, 47, 223, 91, 222, 192, 101, 148, 141, 96, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8, 127, 220]
>>> len(a)
201
```

And 201 - 6 = 195. And 201 is smaller than the maximum allowed 720 bytes (https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/rtk/rtk_impl.cpp#L36). I see you send it as a bytearray. Did you try to send it as a string instead? I suspect that python converts the array to a string representation of the array, which is too long:

```
>>> len("array('B', [211, 0, 195, 67, 80, 0, 139, 71, 123, 38, 0, 0, 41, 64, 136, 2, 0, 0, 0, 0, 32, 0, 128, 0, 91, 210, 156, 162, 136, 142, 162, 140, 142, 0, 0, 0, 20, 148, 195, 103, 118, 35, 211, 231, 84, 126, 161, 237, 144, 4, 127, 52, 252, 180, 10, 104, 52, 188, 98, 14, 56, 249, 67, 131, 126, 5, 238, 216, 70, 95, 130, 173, 62, 54, 203, 251, 173, 167, 40, 165, 236, 36, 132, 12, 172, 94, 12, 149, 44, 4, 153, 149, 224, 219, 177, 224, 16, 161, 249, 42, 218, 0, 109, 230, 27, 143, 58, 212, 69, 145, 99, 217, 22, 69, 145, 18, 210, 160, 0, 37, 11, 194, 208, 128, 50, 11, 130, 144, 112, 26, 12, 156, 28, 188, 185, 78, 28, 9, 240, 47, 223, 91, 222, 192, 101, 148, 141, 96, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 8, 127, 220])")
826
```

xindax2000, 2022-07-30:
I convert it to the string with the str() function.
How should it be done?

dayjaby, 2022-07-30:
Can you try with `a.decode("latin1")` instead of `str(a)`?

xindax2000, 2022-07-31:
This metod (latin1) sends the data, but the RTK doesn't work

julianoes, 2023-06-06:
The MAVSDK-Python Rtk plugin should now work.

This is the [fix on the C++ side](https://github.com/mavlink/MAVSDK/pull/2073).

You need mavsdk_server version [v1.4.16](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.16) and MAVSDK-Python [v1.4.8](https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.8).

Example how to use the Rtk Plugin in Python: [rtk.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/rtk.py).

closed 2023-06-06 by julianoes