mavlink/MAVSDK-Python #496: RaspBerry Pi+ pixhawk 2.4..8 python code is executing in Terminal but drone not taking off after arming

Murali-IsPagro, 2022-08-01:
Hi,

I am testing a Mavsdk python code of offboard position NED. Many times i am observing code is running in terminal but drone is not responding after "arm" command. **Sometimes drone is working as per code**

I am sharing the log file and Terminal output.

https://review.px4.io/plot_app?log=ef670682-193d-4dca-9388-f5c09d180e3f

<sub>pi@raspberrypi:~ $ python3 off_adc_1.py
/home/pi/off_adc_1.py:19: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(4, GPIO.OUT) # output rf
value=1
value=1
value=1
value=1
value=1
Button pressed in run loop
Waiting for drone to have a global position estimate...
GPS info: GpsInfo: [num_satellites: 12, fix_type: FIX_3D]
12
Code Initialized---
Waiting for drone to connect...
Drone discovered!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Fetching orientation
163.06
-- Arming
Status: INFO: Armed by external command	
-- Setting initial setpoint
-- Starting offboard
-- takeoff to 1m</sub>

mrpollo, 2022-08-01:
I sometimes see this same behavior when I try to connect QGC and MAVSDK on the same MAVLink default port

julianoes, 2022-08-02:
Either check for debugging output like this:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

Or create a subscription to telemetry.status_text and check what errors you see.

Murali-IsPagro, 2022-08-02:
> I sometimes see this same behavior when I try to connect QGC and MAVSDK on the same MAVLink default port

When i tried testing this from laptop by opening QGC , The code will automatically make QGC to disconnect and execute the code.

Murali-IsPagro, 2022-08-02:
> Either check for debugging output like this: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues
> 
> Or create a subscription to telemetry.status_text and check what errors you see.

I have already added the status_text in the code. Meanwhile I have attached the Logged messages seen from the log file shared above. I will try running code once again and share the log file and error output, if any. 
[image]

Murali-IsPagro, 2022-08-02:
Hi @julianoes 

As suggested by you, I have tested again and noted that telemetry.status_text showing message till going to offboard mode  ,but once it goes to offboard mode ,it is not displaying status_text. 

>pi@raspberrypi:~ $ python3 off_adc_1.py
value=0
value=0
value=0
value=0
value=1
value=2
value=2
value=2
value=2
in run loop
Waiting for drone to have a global position estimate...
GPS info: GpsInfo: [num_satellites: 11, fix_type: FIX_3D]
11
Code Initialized---
Waiting for drone to connect...
Drone discovered!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Fetching orientation
80.14
-- Arming
_**Status: INFO: Armed by external command**_	
-- Setting initial setpoint
-- Starting offboard
-- takeoff to 1m
value=2

julianoes, 2022-08-02:
It looks like the connection is timing out a couple of times and it fails to get params. Are you sure the mavlink connection is working properly both ways? In the command line you did not specify a baudrate, what baudrate are you using? Is it high enough?

Murali-IsPagro, 2022-08-03:
> It looks like the connection is timing out a couple of times and it fails to get params. Are you sure the mavlink connection is working properly both ways? In the command line you did not specify a baudrate, what baudrate are you using? Is it high enough?

Ah! I will add the baudrate in command line also in the python script and test it. Baud rate is 57600.

Murali-IsPagro, 2022-08-03:
I found the solution. By keeping the drone on ground, I subscribed to odometry, position_body.z_m ,value is showing -3.0 . So after arming and moving to offboard mode, what ever incremental value we gave in "d"(NED). Drone is trying to go down. 

I obtained z value and  added in the code as below:

await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, (z_m-y), 0.0))

where y is the value obtained from potentiometer.

This  solved my problem. Thanks for the community!

julianoes, 2022-08-03:
Ah right, that makes sense. I wouldn't use odometry but rather [position_velocity_ned](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Telemetry.position_velocity_ned) but it should be the same result.

closed 2022-08-03 by julianoes