mavlink/MAVSDK-Python #500: MAVLink for my own Auto-Pilot System

patatessever, 2022-08-05:
Hey, I am wondering if is it possible to use MAVLink to communicate between 2 Jetson Nanos. The idea is that 1 Jetson will be Ground control with QGC and the second will be on my drone. I want to use MAVlink to obtain commands from QGC and on the Jetson which is on the drone, I will receive and manipulate this data for my Auto-Pilot application.  Is that possible, can I just send and receive data from the UDP port of two Jetson Nanos without using ArduPilot like auto-pilot?

julianoes, 2022-08-08:
If you look at the plugins, some are named `..._server`. These are usually meant to run in the air and then talk to a ground station. However, this only works well using C++ for now because of some limitations of the wrappers.

This is currently work in progress and will work better in the future. Would C++ be an option on board?

patatessever, 2022-08-08:
If it is possible in any language  I will be trying. I hoped to use Python however C++ is also good for me. Thank you julianoes.

closed 2022-08-09 by julianoes