mavlink/MAVSDK-Python #502: Install mavsdk on raspbian failed

huytd-pi, 2022-08-08:
My hardware is raspberry pi 4 4gb
```
dronepi@dronepi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
I install mavsdk with command `pip3 install --user mavsdk`
and i am getting error like this when i import the module, help me out

```
dronepi@dronepi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mavsdk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dronepi/.local/lib/python3.7/site-packages/mavsdk/__init__.py", line 7, in <module>
    from .system import System
  File "/home/dronepi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 6, in <module>
    from .async_plugin_manager import AsyncPluginManager
  File "/home/dronepi/.local/lib/python3.7/site-packages/mavsdk/async_plugin_manager.py", line 3, in <module>
    import aiogrpc
  File "/home/dronepi/.local/lib/python3.7/site-packages/aiogrpc/__init__.py", line 1, in <module>
    from grpc import ChannelConnectivity, StatusCode, RpcError, RpcContext,\
  File "/usr/local/lib/python3.7/dist-packages/grpc/__init__.py", line 22, in <module>
    from grpc import _compression
  File "/usr/local/lib/python3.7/dist-packages/grpc/_compression.py", line 15, in <module>
    from grpc._cython import cygrpc
ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by /usr/local/lib/python3.7/dist-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)
```
Thanks!

julianoes, 2022-08-09:
Try this https://github.com/mavlink/MAVSDK-Python/issues/436#issuecomment-1032423478.

huytd-pi, 2022-08-10:
I tried deleting and reinstalling grpcio grpcio-tools, but finally i realized, in my case uninstall grpcio l then the mavsdk module imported successfully

julianoes, 2022-08-10:
Can you write the down the exact steps that worked for you, just for someone seeing this later? Thanks.