mavlink/MAVSDK-Python #503: The example offboard_position_velocity_ned.py in MAVSDK-Python run only if Qgroundcontrol is open?

ioatzi, 2022-08-09:
Hi,
I am use drone.offboard.set_position_ned function to move the drone with cartesian definision and follow a target using camera.

Because, i am use camera stream from the drone, i am not able to use at the same time cv2.imshow with Qgroundcontrol there is a conflict and also the function drone.offboard.set_position_ned  not working if the Qgroundcontrol works.

Anyone has the same issue or can solve my problem. Thank you in advance and please help as I really have no idea how to move forward on this.

Thank you

julianoes, 2022-08-10:
How are you connecting to the drone? If you use the port 14550, then QGC can't at the same time connect to it.

To workaround this you can use something like mavlink-router to split the ports. If you use SITL there is mavlink traffic on ports 14550 as well as 14540, so you can use 14550 for QGC and 14540 for MAVSDK.

ioatzi, 2022-08-10:
Thank you so much for your response!
I am use px4 simulation SITL with gazebo and python script. I use 14540 port to connect with drone with MAVSDK. 
How to config port for QGC and how to run function drone.offboard.set_position_ned without Qgroundcontrol works.

julianoes, 2022-08-10:
> How to config port for QGC

So QGC usually listens automatically on UDP port 14550 if you have "Application settings" -> "General" -> "AutoConnect to the following devices" -> "UDP" checked. That's at least if it's on the same computer, otherwise you have to enable broadcast in the network:
https://docs.px4.io/v1.13/en/advanced_config/parameter_reference.html#MAV_0_BROADCAST (or 1 or 2 whichever mavlink instance it is).

> how to run function drone.offboard.set_position_ned

Have you seen the example?
https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py

And the reference?
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/offboard.html

ioatzi, 2022-08-10:
> > How to config port for QGC
> 
> So QGC usually listens automatically on UDP port 14550 if you have "Application settings" -> "General" -> "AutoConnect to the following devices" -> "UDP" checked. That's at least if it's on the same computer, otherwise you have to enable broadcast in the network: https://docs.px4.io/v1.13/en/advanced_config/parameter_reference.html#MAV_0_BROADCAST (or 1 or 2 whichever mavlink instance it is).
> ok!
> > how to run function drone.offboard.set_position_ned
> 
> Have you seen the example? https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py
> Yes i try this example and works only if QGC is open. If I run without QGC the drone not takeoff, not move and return a error code:  NO_SETPOINT SET 
> And the reference? http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/offboard.html

julianoes, 2022-08-11:
And this code doesn't work? And if so, what is the error? Or what happens?

ioatzi, 2022-08-11:
Work only if QGC is open. If I run without QGC the drone not takeoff, not move and return a error code: NO_SETPOINT SET. My problem is that i want to use this function " uav.offboard.set_position_ned" and this "await uav.offboard.start()"  without QGC.

julianoes, 2022-08-11:
Ok, so we need to understand what the error is.

Make sure to subscribe to telemetry.statustext() to see what messages are sent.

And if you want to see all debug output, try these steps:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

ioatzi, 2022-08-11:
This is mavsdk_server:
[01:17:24|Info ] MAVSDK version: v1.4.4 (mavsdk_impl.cpp:20)
[01:17:24|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:20)
[01:17:44|Info ] New system on: 127.0.0.1:14580 (with sysid: 1) (udp_connection.cpp:192)
[01:17:44|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:482)
[01:17:44|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[01:17:45|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[01:17:45|Info ] System discovered (connection_initiator.h:63)
[01:17:45|Info ] Server started (grpc_server.cpp:53)
[01:17:45|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[01:17:45|Debug] Component Gimbal (154) added. (system_impl.cpp:377)
[01:17:46|Debug] MAVLink: info: [logger] ./log/2022-08-11/10_17_40.ulg	 (system_impl.cpp:250)
[01:17:46|Debug] MAVLink: critical: Attitude invalid (system_impl.cpp:250)
[01:17:46|Debug] MAVLink: critical: Manual control unavailable (system_impl.cpp:250)
[01:17:47|Debug] MAVLink: critical: First waypoint too far away: 5172455m, 900 max	 (system_impl.cpp:250)
[01:18:48|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[01:18:48|Debug] MAVLink: critical: Failsafe enabled: No manual control stick input	 (system_impl.cpp:250)
[01:18:48|Debug] MAVLink: info: Failsafe mode activated	 (system_impl.cpp:250)
[01:18:59|Debug] MAVLink: info: Disarmed by auto preflight disarming	 (system_impl.cpp:250)
[01:19:00|Debug] MAVLink: info: Failsafe mode deactivated	 (system_impl.cpp:250)

This is my console code:
Establishing Connection...
UAV target UUID: {state.uuid}
Establishing GPS lock on UAV..
Established GPS lock...
Fetching amsl altitude at home location....
Flight mode: HOLD
Arming UAV
-- Setting initial setpoint
-- Starting offboard
Now i have no error of offboard function. The drone in the begin takeoff but very quit landing and disarmed.
I change value of this function await uav.offboard.set_position_ned(PositionNedYaw(1.0, 1.0, -5.0, 0.0)) but the drone not move???

julianoes, 2022-08-12:
Try setting the param [COM_RCL_EXCEPT](https://docs.px4.io/v1.13/en/advanced_config/parameter_reference.html#COM_RCL_EXCEPT) to 7 so that not having RC is ignored in offboard mode. I suspect this is why you see:

>[01:18:48|Debug] MAVLink: critical: Failsafe enabled: No manual control stick input (system_impl.cpp:250)
[01:18:48|Debug] MAVLink: info: Failsafe mode activated (system_impl.cpp:250)

ioatzi, 2022-08-12:
Thank you very much its working now!!!

closed 2022-08-15 by JonasVautherin