mavlink/MAVSDK-Python #504: VelocitybodyyawSpeed() - during hover drone is getting yaw.

Murali-IsPagro, 2022-08-12:
Hi,

I am testing my drone using NED code.  I am taking off drone to certain height using VelocityBodyyawSpeed(0.0,0.0,-1.0,0.0) for 5 seconds and hovering for 20 seconds. But i saw drone is getting yaw . I have attached the yaw data from the test below.

>murali@Lenovo:~/MAVSDK-Python/examples$ python3 offboard_velocity_body_test.py
Waiting for drone to connect...
Drone discovered!
-- Fetching orientation
8.27
-- Arming
-- Setting initial setpoint
-- Starting offboard
in here
8.27
8.17
8.120000000000001
8.14
8.23
9.78
12.84
15.4
18.46
22.240000000000002
25.72
27.98
29.05
29.23
hover
29.68
31.77
34.2
35.9
37.410000000000004
40.31
41.1
41.04
40.97
44.12
45.64
45.5
45.25
46.21
45.93
45.07
43.64
43.02
42.89
44.39
46.910000000000004
47.52
47.15
47.22
47.480000000000004
47.83
47.54
46.85
46.94
47.1
46.38
44.300000000000004
42.42
41.64
41.13
40.78
40.99
42.22
44.49
47.01
48.410000000000004
48.39
47.81
47.36
47.4
47.54
47.62
47.24
46.83
46.33
46.29
47.25
48.77
50.44
-- Stopping offboard

This is the code:

>#!/usr/bin/env python3

import asyncio

from mavsdk import System
from mavsdk.offboard import (OffboardError, VelocityBodyYawspeed)

async def run():
    """ Does Offboard control using velocity body coordinates. """

    drone = System()
    await drone.connect(system_address="serial:///dev/ttyUSB0")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break

    print("-- Fetching orientation")
    async for gps_info in drone.telemetry.heading():
        yaw = gps_info.heading_deg
        print(yaw)
        break

    print("-- Arming")
    await drone.action.arm()

    print("-- Setting initial setpoint")
    await drone.offboard.set_velocity_body(
        VelocityBodyYawspeed(0.0, 0.0, 0.0, 0.0))

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: \
              {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        return

    asyncio.ensure_future(yaw_correction(drone)) #subscribing yaw data from telemetry

    print("in here")
    await drone.offboard.set_velocity_body(
        VelocityBodyYawspeed(0.0, 0.0, -1.5, 0.0))
    await asyncio.sleep(6)
    print("hover")
    await drone.offboard.set_velocity_body(
        VelocityBodyYawspeed(0.0, 0.0, 0.0, 0.0))
    await asyncio.sleep(1)

    for a in range(20):
        await drone.offboard.set_velocity_body(
            VelocityBodyYawspeed(0.0, 0.0, 0.0, 0.0))
        await asyncio.sleep(1)

    async for gps_info in drone.telemetry.heading():
        yaw1 = gps_info.heading_deg
        print(yaw1)
        break

    print("-- Stopping offboard")
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: \
              {error._result.result}")

    await drone.action.land()

async def yaw_correction(drone):
    async for gps_info in drone.telemetry.heading():
        yaw = gps_info.heading_deg
        print(yaw)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

julianoes, 2022-08-12:
Ok, when you control a yawspeed, then you probably want to "close the loop" in your code, so then you control yaw by setting a yawspeed to reach it, basically a simple P controller.

Murali-IsPagro, 2022-08-12:
Yes, I agree. But I am not giving any yaw deg/sec command in my code. But still I observe  yaw. Is there any parameters I should change or it is natural for the drone to yaw slightly when VelocityBodyYawspeed() command is running without giving yaw (deg/sec)

julianoes, 2022-08-17:
It is natural that there will be drift, and you need to control that drift by "closing the loop", e.g. by writing a controller to do that.

closed 2022-08-22 by Murali-IsPagro