mavlink/MAVSDK-Python #506: Custom Actuator and Custom Mavlink Messages

alireza787b, 2022-08-16:
Hi good day 
I have a custom actuator which is parachute servo that can be controlled from the board and also some custom uorb topics that will be broadcasted with a custom mavlink message.
I want to get the custom mavlink message within my mavsdk app and can send command to the parachute servo from the app

I know it is possible and we should do it inside the Mavsdk C++ and then auto-generate the functionality in python but couldn't find good resources for that. Can you please recommend some docs on how to do it? Maybe an example or hint.  I will contribute and make the doc for this topic later on :))

JonasVautherin, 2022-08-16:
Doesn't [`action.set_actuator()`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L104) do what you need?

alireza787b, 2022-08-18:
Thanks How could I miss that? it seems it is precisely what I was looking for.
What about custom Mavlink messages or subscribing to specific uORB topics being broadcasted vis mavlink?