mavlink/MAVSDK-Python #508: Unable to parse landing item in QGC plan.

theloni-monk, 2022-08-16:
Hi, I am trying to upload a mission to an sitl instance of the px4 flight simulator so i started by trying to run the mission_import.py example. The example plan provided in the examples directory worked, but was rejected by the simulator with the phrase ```WARN  [navigator] Mission rejected: landing pattern required.``` 
I added the following landing item to the end of the mission items in the .plan file:
```json
{
                "altitudesAreRelative": true,
                "complexItemType": "fwLandingPattern",
                "landCoordinate": [
                    47.397741626132294,
                    8.564924213007533,
                    0
                ],
                "landingApproachCoordinate": [
                    47.397741626132294,
                    8.554924213007533,
                    10
                ],
                "loiterClockwise": true,
                "loiterRadius": 75,
                "stopTakingPhotos": true,
                "stopVideoPhotos": true,
                "type": "ComplexItem",
                "valueSetIsDistance": false,
                "version": 1
            }
```
But I was then met with the error from MAVSDK-Python, 
```
Traceback (most recent call last):
  File "upload_plan.py", line 26, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "upload_plan.py", line 17, in run
    mission_import_data = await drone.mission_raw.import_qgroundcontrol_mission("fp_export.plan")
  File "/net/flood/home/theloni/pymav-expriment-runner/venv/lib/python3.6/site-packages/mavsdk/mission_raw.py", line 973, in import_qgroundcontrol_mission
    raise MissionRawError(result, "import_qgroundcontrol_mission()", qgc_plan_path)
mavsdk.mission_raw.MissionRawError: FAILED_TO_PARSE_QGC_PLAN: 'Failed To Parse Qgc Plan'; origin: import_qgroundcontrol_mission(); params: ('fp_export.plan',)
```
I also have a personal plan I'd like to upload which has the same behavior of only working on the mavsdk python side when its landing item is removed, but being rejected for lack thereof.

julianoes, 2022-08-17:
So the way I understand it is that QGC added this `"complexItemType": "fwLandingPattern",` and we now need to add support for it in MAVSDK, correct? This is not really a MAVLink standard thing but just an extension/feature that QGC added.

Contributions to support it here are welcome:
https://github.com/mavlink/MAVSDK/blob/425577aa5629d591a854a65fd1f2dd6d88b526d9/src/mavsdk/plugins/mission_raw/mission_import.cpp#L172-L219

MyatToe, 2024-03-22:
@julianoes Is there any solutions for mavSDK Python package?

MyatToe, 2024-03-22:
> So the way I understand it is that QGC added this `"complexItemType": "fwLandingPattern",` and we now need to add support for it in MAVSDK, correct? This is not really a MAVLink standard thing but [...]

@julianoes  Is there any solutions for mavSDK Python package?