mavlink/MAVSDK-Python #509: MAVSDK-Python connection not working

dulingch, 2022-08-22:
Issue:  I do not think that I am getting mavsdk to connect to sit from within the environment that is started with the apython command.

Relevant facts:
1) I'm working on a computer running Windows 10.  
2) I'm able to open the jmavsim sitl instance via the run-console.bat using the command make px4_sitl jmavsim.
3) I can command the quadrotor in the sitl instance to takeoff and land from the run-console terminal using with either QGC or if I've double-clicked on mavsdk_server_win32.
4) While the console started double-clicking on mavsdk_server_win32.exe is open, it is continually showing that there's a discovered component and then that heartbeats time out.  The timeout message is showing 3 seconds after the discovered component message.
5)  I'm attempting to start mavsdk through a command prompt using WSL2. I'm running Python 3.7.13 within WSL.   I navigate to a temp folder within the PX4 folder  on my hard drive and activate a virtual environment with the command "python -m venv venv" and then "source ./venv/bin/activate"
6)   I install mavsdk with "pip3 install mavsdk", although at this point, the requirements are satisfied, because mavsdk is in my Python 3.7 site-packages folder.  Then I run the command "pip3 install aioconsole", which is followed by the apython command
7)  at this point, I have the sitl instance open and its associated terminal, the mavsdk_server_win32.exe terminal open that's alternating between a discovered component and heartbeats timing out, and a terminal that's in WSL and a venv that's in an ayncio event loop.  I also have a windows command prompt open as an admin with the "netstat -a" command running in order to monitor established and listening connections.  I'm not seeing any udp connections on the netstat terminal, just TCP.
8)  I enter the commands:
from mavsdk import System
drone = System()
await drone.connect()
in the WSL terminal operating in a venv.  at this point, the WSL terminal just waits and does not provide a ">>>" prompt.  The netstat terminal isn't showing any UDP connections coming up.
9)  for what its worth, if I don't have the mavsdk_server terminal open, run QGC and then open the mavsd_server terminal, I get a bind error on 14540 in the server terminal.  If I am running the mavsdk_server terminal and try to run QGC, then it shows that the default UDP ports are already bound.  
10) after a few minutes of running, the netstat terminal did show a number of UDP ports.  It seems like the most relevant ports, based on IP address were:
 UDP    127.0.0.1:1900         *:*
  UDP    127.0.0.1:54080        *:*
  UDP    127.0.0.1:56165        *:*
  UDP    127.0.0.1:59421        *:*
  UDP    127.0.0.1:59436        *:*
  UDP    127.0.0.1:65449        *:*
That doesn't seem right based on the default settings for MAVSDK.  I also did a control-F search of the mavsdk_server_win32 file and verfied that the defaults are set for 14540.

julianoes, 2022-08-24:
Have you tried these steps? http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

And what connection args are you passing to mavsdk_server_win32?

dulingch, 2022-08-24:
I did run through the debugging steps, but the mavsdk instance that I'm trying to run in the asycio event loop is still not connecting. 
1)  I've tried running the command "await drone.connect(system_address="udp://:14540") and mavsdk_server_win32.exe without input arguments.  
2)  I've tried running the command "await drone.connect() and "./mavsdk_server_win32.exe udp://:14540" 
3)  the mavsdk_server_win32.exe terminal will connect with the jmavsim sitl as the mavsdk_server console that I start does show a connection if I run the sitl, get jmavsim started, and then run the server from powershell using "./mavsdk_server_win32.exe udp://:14540".  

When the jmavsim sitl is open, I have a wsl terminal trying to run an example mavsdk script, and I start mavsdk_server_win32 with the input argument "udp://:14540", mavsdk_server_win32 starts and shows:

- that's v1.4.4
- waiting to discover a system on udp://:14540...
- New System on: 127.0.0.1:14580 (with sysid: 1)
- New: System ID: 1 Comp ID: 1 added.
- Discovered 1 component(s)
- Server started
- Server set to listen on 0.0.0.0:50051
- MAVLink: Info: Data link regained
and then it alternates between timing out and discovering a component

Meanwhile, the terminal where I just entered the command "await drone.connect()" just sits with a blank line after the command is entered (I did hit enter). 

What is the difference between the mavsdk_server_win32.exe file (9.6 Mb) and the mavsdk_server (36Mb) file that is build in the /mavsdk/bin folder with the command "python3 setup.py bdist_wheel", aside from the latter file being about 4x as large as the executable?  Not sure how to run the latter file, as I get errors if I attempt to run it as a python file.

I appreciate your help with this Julian.  I'm not sure if there's a problem here that's related to how the server is alternating between discovering and timing out, but that's not preventing me from running commander takeoff and commander land from the sitl terminal.   The server terminal shows messages that reflect the takeoff and landing done within the sitl, too.  There's just no connection between the terminal in the asyncio event loop and the server and sitl terminals.

I apologize for confusing the use of terminal, console, shell, etc...  I'm an aerospace engineer.

JonasVautherin, 2022-08-24:
```python
from mavsdk import System
drone = System()
await drone.connect()
```

This should run the embedded `mavsdk_server`, and if I understand correctly, you are already running it manually.

Can you try the following instead?

```python
from mavsdk import System
drone = System(mavsdk_server_address="localhost")
await drone.connect()
```

dulingch, 2022-08-26:
I uninstalled WSL2 and reinstalled it to clean up directories with python, because I had previously been using pyenv and I thought that may have been causing a problem.

I reinstalled Python 3 (its 3.8) and then went through quickstart for mavsdk.  still no connection out of the terminal where I start the asyncio event loop.  I've tried a couple ways to get mavsdk_server running and I can at least confirm that the drone.connect command in the asyncio loop is binding udp ports, because when I ran mavsdk_server from a CLI, it reported a bind error that the udp port was already in use.  

At this point, I'm going to: 1) talk to my IT support to see if there are any firewalls that could be preventing the udp port connection, 2) try it on another laptop, 3)  and try to connect via TCP (I know there's at least one issue that I can check for guidance on that).  I'll get back to you after this.  This is ridiculous.  I know that your code works.  I'll post again after I go through those actions, so that this doesn't waste more of your time trying to figure this out.

julianoes, 2022-08-26:
> What is the difference between the mavsdk_server_win32.exe file (9.6 Mb) and the mavsdk_server (36Mb) file that is build in the /mavsdk/bin folder with the command "python3 setup.py bdist_wheel", [...]

Interesting. It could be that the bigger one is the debug build with all debug symbols. I have to check that.

@dulingch please, when you get output from mavsdk_server paste the full output here. That might give me clues. And then also paste in the Python example that you are trying.

dulingch, 2022-08-26:
I quit trying to use the Cygwin-based toolchain and am running headless with the WSL2 development environment.  Installation was easy and connection was immediate.

dulingch, 2022-08-26:
The WSL2 environment worked immediately.  Thanks for taking the time to help.

closed 2022-08-26 by dulingch