mavlink/MAVSDK-Python #513: MAVSDK UBUNTU

xindax2000, 2022-09-18:
I want to join the Pixhawk with MAVsdk. 
I use the following command:
./mavsdk_server -p 50051 serial:///dev/ttyACM1:57600
=> does NOT start... just waiting
If I open and close QGroundControl, than It works!!!
If I don't open QGroundControl, then it won't start.

What schould I do?
How do I start it?

dayjaby, 2022-09-18:
You have to disable any kind of autoconnect on the serial ports in QGroundControl first.

The best method to find the culprit in cases like this is to use lsof:

```
sudo apt install lsof
sudo lsof /dev/ttyACM0
```

JonasVautherin, 2022-09-18:
> I want to join the Pixhawk with MAVsdk.

PX4 or Ardupilot? I'm assuming PX4.

> If I open and close QGroundControl, than It works!!!

I do believe that over serial, PX4 needs to receive MAVLink messages before it starts sending them. I think this is fairly new, though. I guess `mavsdk_server` should automatically send heartbeats when connecting over serial. Not sure... @julianoes do you have an opinion here?

xindax2000, 2022-09-18:
Yes, PX4
how can i cause heartbeat?

JonasVautherin, 2022-09-18:
I think you would want to set [`always_send_heartbeats`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/include/mavsdk/mavsdk.h#L200), using [`set_configuration`](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/include/mavsdk/mavsdk.h#L270) maybe somewhere around [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/connection_initiator.h#L22)?

julianoes, 2022-09-18:
@JonasVautherin should we make a bugfix release for this?

JonasVautherin, 2022-09-18:
I guess it would be nice to test it first. Also we should do it only for `serial://`, shouldn't we?

xindax2000, 2022-09-18:
Yes, PLS!

julianoes, 2022-09-18:
@JonasVautherin, yes but it's not straightforward I'm afraid.

xindax2000, 2022-10-16:
Did you manage to solve it?

julianoes, 2023-02-10:
This should be fixed in https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.4.

closed 2023-02-10 by julianoes