mavlink/MAVSDK-Python #514: sitl_multiple_run.sh

dulingch, 2022-09-22:
Can the multiple aircraft sitl be run in a headless mode with jmavsim or gazebo?  

I'm using MAVSDK python in Ubuntu on a Windows computer.  I installed PX4 through the WSL2 toolchain.  Based on what I have read on the website for MAVSDK, headless is the only way to go with the WSL2 installation.  

Here's what's happening.  I am trying to run the sitl_multiple_run.sh shell script in an Ubuntu console for two aircraft.  The computer is able to start two instances in the PX4-Autopilot/Tools/build/px4_sitl_default directory.  When I run jmavsim_run.sh for instance 0 or instance 1, the is an exception that comes up:

Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:567)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.awt.HeadlessException:
No X11 DISPLAY variable was set,
but this program performed an operation which requires it.
        at java.desktop/java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:165)
        at java.desktop/java.awt.Window.<init>(Window.java:538)
        at java.desktop/java.awt.Frame.<init>(Frame.java:423)
        at java.desktop/java.awt.Frame.<init>(Frame.java:388)
        at java.desktop/javax.swing.JFrame.<init>(JFrame.java:180)
        at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:107)
        at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
        at me.drton.jmavsim.Simulator.main(Simulator.java:944)
        ... 5 more

Is this even a viable option for executing multiple aircraft simulation in a SITL environment?

I'm sorting through a few other issue threads related to multiple aircraft SITL right now.

Thank you,

Chris Duling

JonasVautherin, 2022-09-22:
```
Starting GUI...
[...]
Caused by: java.awt.HeadlessException:
No X11 DISPLAY variable was set,
```

What makes you say you are running it headless? Seems like you are not, doesn't it? :thinking:

dulingch, 2022-09-25:
I'm not sure; however, we've been able to get the functionality that we need by using pymavlink and Mission Planner to work with two aircraft simultaneously,  With that said, I am going to call my issue closed.

closed 2022-09-25 by dulingch

dulingch, 2022-10-11:
I assume that it was running headless, be a use a jmavsim view of the
aircraft did not come up.  Additionally, I'm under the assumption that the
WSL2 tool chain installed on a windows 10 machine is only able to operate
without the jmavsim or gazebo visualization of the aircraft.  I've had no
issues running the headless implementation of PX4 with a single aircraft,
but am trying to work with multiple aircraft now.