mavlink/MAVSDK-Python #516: Make maneuvers along mission path

YasinBedirhanSimsek, 2022-09-27:
Hi everyone, 

Is it possible to perform sudden roll-pitch-yaw maneuvers during a mission but not stop or pause the mission. (For example dodging projectiles)

Is there any way I can do this smoothly ?

I tried the offboard mode but I have to pause the mission, perform the maneuvers and resume it.

julianoes, 2022-09-27:
You would have to offload the full mission to offboard to do that. Or you could try to use the avoidance interface. Have a look at https://github.com/PX4/PX4-Avoidance. I'm not sure how maintained this is though but the MAVLink interface that it uses might give you an idea.

There is also this feature request around this topic: https://github.com/mavlink/MAVSDK/issues/1665

YasinBedirhanSimsek, 2022-09-27:
Thank you for your response, i have been working on the offboard mode but there is problem with it. If i can solve it everything will be perfect.

Think of a very basic example:

1-) Drone arms and takes off to 5m
2-) I get the current gps coordinates when it reaches 5m
3-) After that I create a very basic mission. First waypoint is the current location of the drone, second one is an offset added to the longitude value. It basically goes straight 
4-) I upload the mission and start it
5-) After 3 seconds, i directly go to offboard mode (not pause or anything just directly start offboard mode). Set some velocity setpoint for going right that lasts 0.5 seconds. The drone performs a roll and goes right.
6-) I switch back to mission mode

Now I expect that drone will continue from where it is. When i pause mission from QGC it does that but not with mavsdk. 

Drone doesn't continue the mission from where it currently is after the roll. It performs a yaw manuever, goes back to the straight line it was following before the offboard mode. When it reaches the line it performs another yaw rotation and starts following the path.

Is it a feature of QGC or is there problem with something else ?

julianoes, 2022-09-28:
Yer, I think that's unfortunately a PX4 issue that's not that easy to get rid of.

YasinBedirhanSimsek, 2022-09-29:
For anyone who might come across the same problem:

It is possible to send the drone to the next waypoint by sending it in offboard mode with GPS setpoint and continue the mission from there.

Drone can continue from the position it is currently at if the GPS go command is given in offboard mode. After that you can get back to mission mode to resume the mission.

closed 2022-09-29 by YasinBedirhanSimsek