mavlink/MAVSDK-Python #517: How to adjust landing or takeoff speed?

jaekyoungkwon, 2022-09-27:
Hello
I'm trying to set landing speed.
I upload 6 mission items and checked the flight mode changed to landing state after all the missions.
But it seems so slowly when it's landing.
I want to change landing speed but I have no idea how to change the landing speed.
Which code do I have to put in this code?

'''
if await drone.mission.is_mission_finished():

    print("-- Landing")
    await drone.action.land()
    await termination_task
    print("-- Landed")
'''

JonasVautherin, 2022-09-27:
Isn't that a PX4 param?

julianoes, 2022-09-27:
If it's a PX4 param, then you can set that param using QGroundControl, or as part of your script using the param plugin.

jaekyoungkwon, 2022-09-28:
[image]

I've added this code the above codes:
    await drone.param.set_param_float("MPC_Z_VEL_MAX_DN", 4.0)

But still I can see the velocity of down is 1.0m/s

jaekyoungkwon, 2022-09-28:
Oh I found the answer!
The Parameter what I have to change is 'MPC_Z_V_AUTO_DN'
I changed code like:

    await drone.param.set_param_float("MPC_Z_V_AUTO_DN", 4.0)

and I checked the landing speed is 4m/s
Thank you!

closed 2022-09-28 by jaekyoungkwon