mavlink/MAVSDK-Python #522: CameraServer plugin "UNIMPLEMENTED"

carlsondev, 2022-10-10:
When running the attached code and connecting to the headless px4 docker container (`jonasvautherin/px4-gazebo-headless`) I receive this error. 
```
Traceback (most recent call last):
  File "/Users/fibonacci/Desktop/Mines/Fall-2022/Field-Session/MAVSDK-Python/custom_examples/camera_server.py", line 20, in <module>
    loop.run_until_complete(run())
  File "/usr/local/Cellar/python@3.10/3.10.7/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/Users/fibonacci/Desktop/Mines/Fall-2022/Field-Session/MAVSDK-Python/custom_examples/camera_server.py", line 13, in run
    await drone.camera_server.set_information(info)
  File "/usr/local/lib/python3.10/site-packages/mavsdk/camera_server.py", line 855, in set_information
    response = await self._stub.SetInformation(request)
  File "/usr/local/lib/python3.10/site-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::1%5D:50051 {grpc_message:"", grpc_status:12, created_time:"2022-10-10T14:00:21.718988-06:00"}"
>
```

Based on a prior reported issue (#394), I think this might be some simple issue with the library.

## Code

```python
import asyncio
from mavsdk import System
from mavsdk.camera_server import Information

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    info = Information("test_vendor", "test_model", "", 30, 30, 30, 1024, 1024, 0, 1, "http://")

    await drone.camera_server.set_information(info)

if __name__ == "__main__":
    # Start the main function
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(run())
```

julianoes, 2022-10-10:
Server plugins haven't been properly verified in the language wrappers as far as I remember. Sorry this is a bit in flux.

JonasVautherin, 2022-10-11:
Are you sure that `mavsdk_server` is running? For instance, can you receive telemetry and run actions?

What version of MAVSDK are you using? Did you build it from source?

It seems to complain that this function you are calling (`set_information`) is not implemented on the server side. But the server should be enabled [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/grpc_server.h#L37) and [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/CMakeLists.txt#L62), and `set_information` is implemented [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/CMakeLists.txt#L62).

julianoes, 2022-10-12:
Let me ask a different question: what are you trying to do?

carlsondev, 2022-10-12:
So, I know that the CameraServer really just receives and responds to events from the Camera plugin. I initially ran into this issue when attempting to subscribe to `take_photo` in my own code and then figured out that it was a broader issue and that it would not be easy to demonstrate if my example code used that function, so I used the `set_information` function (which also yields that error). All the other plugins (except the ones I have created issues for) work just fine, this one does not.

julianoes, 2022-10-18:
Right, I guess we can add it then. I'm sorry I'm a bit swamped right now, so if you can make pull requests and test things I can merge things but I can't push stuff forward much at this point.

carlsondev, 2022-10-19:
No problem. Can you point me in the right direction since the C++ and Python grpc stuff appears to match pretty well to my eye

julianoes, 2022-10-24:
I wonder if this is due to the changes that have gone in towards v2 of MAVSDK, and the Python wrappers are still against v1.4.

JonasVautherin, 2022-10-24:
But we haven't changed the proto files, so the data going over protobuf should be the same, right?

carlsondev, 2022-10-25:
Are you guys able to replicate my issue using the provided example?

GeoSpark, 2023-07-27:
Has there been any movement on this? I wouldn't mind getting my hands dirty, but I'm not sure of the best place to start. Like @carlsondev I am trying to run a camera server:
```
import asyncio

from mavsdk import System

async def main():
    print('Waiting to connect...')
    drone = System()
    await drone.connect('serial:///dev/ttyAMA0:115200')

    async for state in drone.core.connection_state():
        if state.is_connected:
            print('...connected')
            break

    asyncio.ensure_future(handle_photo(drone))

    while True:
        await asyncio.sleep(1)

async def handle_photo(drone):
    async for photo in drone.camera_server.take_photo():
        print('Photo taken')

if __name__ == '__main__':
    asyncio.run(main())
```

And I get the same sort of error:
```
Waiting to connect...
...connected
__anext__ grpc exception
Traceback (most recent call last):
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/grpc/_channel.py", line 864, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNIMPLEMENTED
        details = ""
        debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-07-27T16:35:26.073212419+01:00", grpc_status:12, grpc_message:""}"
>
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<handle_photo() done, defined at /home/merseyviking/development/pymavsdk-camera/main.py:21> exception=<_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNIMPLEMENTED
        details = ""
        debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-07-27T16:35:26.073212419+01:00", grpc_status:12, grpc_message:""}"
>>
Traceback (most recent call last):
  File "/home/merseyviking/development/pymavsdk-camera/main.py", line 22, in handle_photo
    async for photo in drone.camera_server.take_photo():
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/mavsdk/camera_server.py", line 905, in take_photo
    async for response in take_photo_stream:
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 154, in __anext__
    return await asyncio.shield(self._next_future)
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/home/merseyviking/development/pymavsdk-camera/venv/lib/python3.9/site-packages/grpc/_channel.py", line 864, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNIMPLEMENTED
        details = ""
        debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-07-27T16:35:26.073212419+01:00", grpc_status:12, grpc_message:""}"
>
```

JonasVautherin, 2023-07-27:
Hmm I would say that the gRPC error is thrown by:

```
    async for photo in drone.camera_server.take_photo():
```

Presumably because your Python version knows the `take_photo()` function, but your `mavsdk_server` backend does not. How did you install MAVSDK and which version is it?

GeoSpark, 2023-07-27:
I installed it in a virtualenv with `pip install mavsdk` The server is running from within there automatically when I create the system with no parameters. I have tried running the server from the command line:
```
merseyviking@raspberrypi:~/development/pymavsdk-camera$ venv/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server  -p 50051 --sysid 245 --compid 190 serial:///dev/ttyAMA0:115200
[06:13:49|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[06:13:49|Info ] Waiting to discover system on serial:///dev/ttyAMA0:115200... (connection_initiator.h:20)
[06:13:49|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[06:13:49|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[06:13:49|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[06:13:49|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[06:13:49|Info ] System discovered (connection_initiator.h:63)
[06:13:49|Info ] Server started (grpc_server.cpp:53)
[06:13:49|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
```

And then connecting to it in my code to see if I can get any useful log info, but nothing gets logged, and it falls over at the same place as before.
```
async def main():
    print('Waiting to connect...')
    drone = System(mavsdk_server_address='localhost')
    await drone.connect('serial:///dev/ttyAMA0:115200')

    async for state in drone.core.connection_state():
        if state.is_connected:
            print('...connected')
            break

    asyncio.ensure_future(handle_photo(drone))

    while True:
        await asyncio.sleep(1)
```

Like @carlsondev I have tried with `set_information()` and again, it reports undefined for that call as per the OP.

JonasVautherin, 2023-07-27:
I don't think `camera_server` is present in `mavsdk_server` v1.4.16: https://github.com/mavlink/MAVSDK/blob/v1.4.16/src/mavsdk_server/src/grpc_server.h. The camera server is just not in the [proto](https://github.com/mavlink/MAVSDK-Proto/tree/17d707685a8cbe5e0b6c930850b5a2c75b85bbd8/protos). So I am a bit surprised to see that the corresponding MAVSDK-Python knows `drone.camera_server` :thinking:. I guess there is a mismatch with the protos?

The `camera_server` is in [main](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/grpc_server.h#L37-L40), though.

Now the thing is that the current `main` will eventually be released as `v2.0.0`, but it's not planned yet. It is being tracked here, though: https://github.com/mavlink/MAVSDK/issues/2002. I am not sure how much work it would take for you to backport the camera server into `v1.4.x` (which we would then have to release as `v1.5.0`). Possibly not completely straightforward.

So realistically, I think your two best solution is to build MAVSDK from the `main` branch until v2 gets released.

@julianoes: or do you think backporting would be easier?

GeoSpark, 2023-07-27:
Thanks for looking into this @JonasVautherin Yeah, it's funny there was a mismatch, but I am glad we have got to the bottom of it. I appreciate the amount of work this is for you, @julianoes and contributors.
I had looked at the server source and saw the camera_server referenced there, but I didn't think to actually look at the 1.4.16 tag :facepalm: 
I'll go ahead and build the latest main branch version for now.

GeoSpark, 2023-07-27:
I can confirm that the latest mavsdk_server v1.4.0-511-gfadfe22e works.

JonasVautherin, 2023-07-27:
Ok let's close this then, I think we have our answer now :blush:. Feel free to re-open if necessary.

closed 2023-07-27 by JonasVautherin