mavlink/MAVSDK-Python #524: telemetry.position() and telemetry.position_velocity_ned() not working

PRouco, 2022-10-23:
When trying the example telemetry.py, the telemetry.position() doesn't seem to be working on a Pixhawk 4, which is exactly what was happening on #176. The controller runs PX4 version v1.13.1 and the plarform is Windows with MAVSDK-Python v1.4.1 and MAVSDK v1.4.7. This was also used:

`./mavsdk_server_bin  serial://COM10:57600`

Some of the rest of the telemetry works as intended, like telemetry.battery(), telemetry.gps_info() or euler_angles(). Like the position, position_velocity_ned() also didn't work. The pixhawk had a GPS fix and the health.is_global_position_ok() gave true as answer. In QGroundControl the position can be seen without problems.

julianoes, 2022-10-24:
Interesting, thanks for the note. I'll try to have a look later this week.

julianoes, 2022-10-28:
Sorry, I didn't get to this yet. 

Have you tried setting the rate with this?
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Telemetry.set_rate_position

PRouco, 2022-10-28:
I'ver tried to to that and doesn't make any difference. The command is executed correctly, but it keeps getting stuck at drone.telemetry.position().

Also, I've done some testing and with the PX4 Stable Release 1.10.0 the telemetry.position_velocity_ned() works correctly, not so the position.

julianoes, 2022-11-02:
Right, so something is broken...

arpitpara, 2023-03-26:
`exception calling callback for <Future at 0x7f0eac55e3d0 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/arpit/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 426, in __next__
    return self._next()
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 826, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
exception calling callback for <Future at 0x7f0eac5536d0 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/arpit/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 426, in __next__
    return self._next()
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 826, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
exception calling callback for <Future at 0x7f0eac564100 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/arpit/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 426, in __next__
    return self._next()
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 826, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
exception calling callback for <Future at 0x7f0eac5c86d0 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/arpit/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 426, in __next__
    return self._next()
  File "/home/arpit/.local/lib/python3.8/site-packages/grpc/_channel.py", line 826, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed`

This is the error I got!

julianoes, 2023-03-27:
@arpitpara I think that's a separate issue but nevertheless something to fix.

I've created an issue: https://github.com/mavlink/MAVSDK-Python/issues/567

julianoes, 2023-03-28:
Regarding the original issue. I can reproduce this with a GPS indoor. The GPS signal is good enough to show up in QGC, however, the GPS signal is in fact too weak to be sent out.

The check happens here in PX4:
https://github.com/PX4/PX4-Autopilot/blob/release/1.13/src/modules/mavlink/streams/GLOBAL_POSITION_INT.hpp#L71

Once, these pass on the PX4 side with proper GPS signal, messages start flowing for me.

Depending on the link, setting the rate might additionally be required, like this:

```
diff --git a/examples/telemetry.py b/examples/telemetry.py
index 7fa831f..318afbe 100755
--- a/examples/telemetry.py
+++ b/examples/telemetry.py
@@ -35,6 +35,8 @@ async def print_in_air(drone):
 
 
 async def print_position(drone):
+    await drone.telemetry.set_rate_position(5)
+
     async for position in drone.telemetry.position():
         print(position)
```

I'm closing this issue. Reopen if none of what I said works.

closed 2023-03-28 by julianoes