mavlink/MAVSDK-Python #525: How to takeoff and land with fixedwing mode?

jaekyoungkwon, 2022-10-24:
Hi

I can do takeoff and land with MultiCopter(MC) mode when I use those code:

    await drone.action.takeoff() # To Takeoff.
    await drone.action.land()     # To Land.

But I just want to do that with FixedWing(FW) mode to takeoff and land.

I tried to transfer to fixedwing mode before takeoff like below:

    await drone.action.transition_to_fixedwing()
    
    print("-- Taking Off")
    await drone.action.takeoff()

But it occured error.
Also same in landing mode:

    await drone.action.transition_to_fixedwing()
    
    print("-- Landing")
    await drone.action.land()

What should I do if I want to fixedwing mode to takeoff and land?

jaekyoungkwon, 2022-10-24:
I’m not sure how to understand this code, but i would say it doesn’t work.

JonasVautherin, 2022-10-24:
So you have a VTOL, but you want to use it as a fixedwing. Do I understand correctly?

julianoes, 2022-10-24:
You can't use transition. However, I think that just sending the takeoff command does not work for fixedwings indeed, at least for PX4. I think the only way is to use manual control or a mission with takeoff item.

jaekyoungkwon, 2022-10-25:
@JonasVautherin Yes exactly right.

jaekyoungkwon, 2022-10-25:
@julianoes How can I do a mission with takeoff item?

julianoes, 2022-11-02:
You have to create the mission yourself with the mission items or alternatively use QGC to create it and then use the mission_raw::import functionality.

jaekyoungkwon, 2022-11-03:
@julianoes I already created missions and uploaded them.
    mission_items.append(MissionItem(47.398170327054470,
                                                             8.545649021863966,
                                                             500,
                                                             False,
                                                             20,                                                             
                                                             60,
                                                             MissionItem.CameraAction.NONE,
                                                             float('nan')
                                                             float('nan')
                                                             float('nan')
                                                             float('nan')
                                                             float('nan')))
    
    mission_plan = MissionPlan(mission_items)
    await drone.mission.upload_mission(mission_plan)

And I found MAVSDK does not support fixedwing takeoff.
How about fixedwing landing using MAVSDK? Is it possible to do that?

julianoes, 2022-11-07:
It's currently not supported. I suggest to use MissionRaw.

jaekyoungkwon, 2022-11-08:
@julianoes I tried to use MissionRaw, but I can see MissionRawError all the time.
I don't know why this error keep occurs. Is there any wrong in my code?

I made list for mission_raw.MissionItem

    mission_items.append(mission_raw.MissionItem(
        1,
        6,
        16,
        0,
        1,
        0, 10, 0, float('nan'),
        int(47.40271757 * 10**7),
        int(8.542850271 * 10**7),
        30.0,
        0
    ))

    mission_items.append(mission_raw.MissionItem(
        1,
        6,
        16,
        0,
        1,
        0, 10, 0, float('nan'),
        int(47.40261559 * 10**7),
        int(8.539117889 * 10**7),
        30.0,
        0
    ))

    await drone.mission_raw.upload_mission(mission_items)

This is the error message
[image]

julianoes, 2022-11-09:
Your sequence (first field) is 1 for both items. It should be 0, 1, ...

jaekyoungkwon, 2022-11-10:
I've already tried that before, but same error occured.

julianoes, 2022-11-11:
If you create a full minimal example for me, then I can try to reproduce it.

jaekyoungkwon, 2022-11-11:
Here is my example:

    from mavsdk import System
    from mavsdk.mission import (MissionItem, MissionPlan)
    from mavsdk.mission_raw import MissionItem
    from mavsdk import mission_raw

    async def px4_connect_drone():
        await asyncio.sleep(2)
        await drone.connect(system_address="udp://:14540)

        print("waiting for drone to connect...")
        async for state in drone.core.connection_state():
            if state.is_connected:
                print("-- Connected to drone!")
                break

    async def run():
        await px4_connect_drone()
        await run_drone()

    async def run_drone():
        # Start parallel tasks
        print_altitude_task = asyncio.ensure_future(print_altitude(drone))
        print_flight_mode_task = asyncio.ensure_future(print_flight_mode(drone))
        print_velocity_task = asyncio.ensure_future(print_velocity(drone))
        print_euler_angle_task = asyncio.ensure_future(print_euler_angle(drone))
        print_mission_progress_task = asyncio.ensure_future(print_mission_progress(drone))

        running_tasks = [print_altitude_task, print_flight_mode_task, print_velocity_task, print_euler_angle_task, print_mission_progress_task ]
        termination_task = asyncio.ensure_future(observe_is_in_air(drone, running_tasks)

        mission_items = []

        mission_items.append(mission_raw.MissionItem(
             1,
             6,
             16,
             0,
             1,
             0, 10, 0, float('nan'),
             int(47.40271757 * 10**7),
             int(8.542850271 * 10**7),
             30.0,
             0
         ))

        mission_items.append(mission_raw.MissionItem(
            1,
            6,
            16,
            0,
            1,
            0, 10, 0, float('nan'),
            int(47.40261559 * 10**7),
            int(8.539117889 * 10**7),
            30.0,
            0
        ))
        
        print("-- Uploadming mission")
        await drone.mission_raw.upload_mission(mission_items)

    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

julianoes, 2022-11-14:
The problem was that none of your items had `current` set to 1. The fist one needs to have current set to 1.

Here is the working minimal example:

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System
from mavsdk import mission_raw

async def px4_connect_drone():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            return drone

async def run():
    drone = await px4_connect_drone()
    await run_drone(drone)

async def run_drone(drone):

    mission_items = []

    mission_items.append(mission_raw.MissionItem(
         0,
         6,
         16,
         1, # FIX: Current!
         1,
         0, 10, 0, float('nan'),
         int(47.40271757 * 10**7),
         int(8.542850271 * 10**7),
         30.0,
         0
     ))

    mission_items.append(mission_raw.MissionItem(
        1,
        6,
        16,
        0,
        1,
        0, 10, 0, float('nan'),
        int(47.40261559 * 10**7),
        int(8.539117889 * 10**7),
        30.0,
        0
    ))

    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)
    print("-- Done")

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

```

closed 2022-11-14 by julianoes

jaekyoungkwon, 2022-11-14:
@julianoes Oh it works! I appreciate that make an example using MissionRaw. :)