mavlink/MAVSDK-Python #527: Detect mission upload or mission start from QGC ?

YasinBedirhanSimsek, 2022-10-31:
What would be the best way to detect (with mavsdk) when the user uploads or starts a mission manually over QGC ? 

I have implemented my own mission navigation using mavsdk.offboard module so I have to switch to offboard mode when user uploads and starts the mission on QGC

JonasVautherin, 2022-10-31:
Not sure I understand what you want. So when the user changes mode to "Mission" from QGC, you would want to intercept that and instead put PX4 in offboard mode?

YasinBedirhanSimsek, 2022-10-31:
My companion computer will listen to mission changes (uploading, clearing, starting etc...) that will come from QGC and react according to that.

For example:

- User will open the QGC on remote ground station and will create a mission with waypoints. 
- When the user uploads that mission to the PX4 via QGC, I want my companion computer to intercept that upload process.

julianoes, 2022-10-31:
The mission_changed API doesn't necessarily work if the MISSION_ACK message doesn't actually end up in MAVSDK. It's a hack and not made on top of a proper API.

One such proper API would have been MISSION_CHANGED but that was rejected for a better MISSION_CHECKSUM.

More info: https://github.com/mavlink/mavlink/pull/1877