mavlink/MAVSDK-Python #530: VisionPositionEstimate

Shorlaks, 2022-11-13:
Hi

according to https://mavlink.io/en/messages/common.html#VISION_POSITION_ESTIMATE
`VisionPositionEstimate` struct has a `reset_counter`.

It was implemented in mavsdk v0.21.1 but not anymore.
This feature is not implemented anymore by design or implemented elsewhere?  

Thanks, Dan

julianoes, 2022-11-13:
Isn't it here? http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mocap.html#mavsdk.mocap.Mocap.set_vision_position_estimate

Shorlaks, 2022-11-14:
@julianoes Hi

`set_vision_position_estimate()` takes `VisionPositionEstimate` object.
Currently `VisionPositionEstimate` takes 4 parameters:
- time_usec (`uint64_t`)
- position_body (`PositionBody`)
- angle_body (`AngleBody`)
- pose_covariance (`Covariance`)

The MAVLink message `VISION_POSITION_ESTIMATE` is built from those parameters.
It supposed to take another parameter:
- reset_counter (`uint_8`)  - Estimate reset counter. This should be incremented when the estimate resets in any of the dimensions (position, velocity, attitude, angular speed). This is designed to be used when e.g an external SLAM system detects a loop-closure and the estimate jumps.

julianoes, 2022-11-17:
Funny, looks like that's just set to 0:

https://github.com/mavlink/MAVSDK/blob/5ec5fcf3e5b2e25e339fac77c548b521f18d4002/src/mavsdk/plugins/mocap/mocap_impl.cpp#L106

Pull requests welcome, first step would be to add it to:
https://github.com/mavlink/MAVSDK-Proto/blob/ee13b7b6386a709618db0d64efbe27529f94cc3d/protos/mocap/mocap.proto#L105