mavlink/MAVSDK-Python #534: Issue downloading ulogs

djlisko01, 2022-12-09:
I'm having an issue with downloading logs after a mission completes.  I ran the code example posted [here](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/logfile_download.p,y) and I get the following error.

```
File "/Users/daniellisko/GitRepos/Px4-Batch-Simulation/missions/ex_log.py", line 21, in run await download_log(drone, entry)
File "/Users/daniellisko/GitRepos/Px4-Batch-Simulation/missions/ex_log.py", line 29, in download_log
async for progress in drone.log_files.download_log_file(entry, filename):
TypeError: 'async for' requires an object with __aiter__ method, got coroutine
```

There are log files because I get the following output:
```
-- Connected to drone!
Log 0 from 2022-12-09T22:54:45Z
Log 1 from 2022-12-09T23:33:25Z
Log 2 from 2022-12-09T23:01:56Z
Log 3 from 2022-12-09T23:25:22Z
Log 4 from 2022-12-09T22:56:17Z
```

I tried this too:
```
 for progress in await drone.log_files.download_log_file(entry, filename):
        new_progress = round(progress.progress*100)
        if new_progress != previous_progress:
            sys.stdout.write(f"\r{new_progress} %")
            sys.stdout.flush()
            previous_progress = new_progress
```

However, I get this error now:

```
 raise LogFilesError(result, "download_log_file()", entry, path)
mavsdk.log_files.LogFilesError: INVALID_ARGUMENT: 'Invalid Argument'; origin: download_log_file(); params: (<mavsdk.log_files.Entry object at 0x10954b490>, './log-2022-12-09T22-54-45Z.ulog')
```

julianoes, 2022-12-11:
Thanks for the issue. I can reproduce this.

```
/home/julianoes/src/upstream/MAVSDK-Python/examples/./logfile_download.py:45: DeprecationWarning: There is no current event loop
  asyncio.get_event_loop().run_until_complete(run())
Waiting for drone to connect...
-- Connected to drone!
Log 0 from 2022-12-01T02:53:10Z
Downloading: log 0 from 2022-12-01T02:53:10Z to ./log-2022-12-01T02-53-10Z.ulog
/home/julianoes/src/upstream/MAVSDK-Python/examples/./logfile_download.py:28: RuntimeWarning: coroutine 'LogFiles.download_log_file' was never awaited
  async for progress in drone.log_files.download_log_file(entry, filename):
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Traceback (most recent call last):
  File "/home/julianoes/src/upstream/MAVSDK-Python/examples/./logfile_download.py", line 45, in <module>
    asyncio.get_event_loop().run_until_complete(run())
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/julianoes/src/upstream/MAVSDK-Python/examples/./logfile_download.py", line 20, in run
    await download_log(drone, entry)
  File "/home/julianoes/src/upstream/MAVSDK-Python/examples/./logfile_download.py", line 28, in download_log
    async for progress in drone.log_files.download_log_file(entry, filename):
TypeError: 'async for' requires an object with __aiter__ method, got coroutine
```

julianoes, 2022-12-11:
@JonasVautherin we seem to have this function `download_log_files` twice:

https://github.com/mavlink/MAVSDK-Python/blob/677a2958d953ddf532fcb6ac4e8c73a4c3f35a58/mavsdk/log_files.py#L381-L477

This must be related to [this PR](https://github.com/mavlink/MAVSDK-Proto/pull/267) but I'm not quite sure why it's still (or now) wrong?

JonasVautherin, 2022-12-12:
I'd be happy if we could remove the second one, as it creates issues with MAVSDK-Java :innocent:

julianoes, 2022-12-13:
Thanks @dayjaby! Unfortunately, this is a bit of a pain to backport to v1.4 but I guess I should do it.

closed 2022-12-13 by julianoes