mavlink/MAVSDK-Python #535: MAVSDK-Python  states that it is only available for Python >= 3.6

Nomed73, 2022-12-20:
I have Python 3.10 installed and I am getting this message while running the examples. A few weeks I was able to run these examples without this issue. Today, I was trying some of the examples and started getting this message. 

[!] MAVSDK-Python is only available on Python >= 3.6

What is causing this and how can I fix this? Any help is appreciated. 

Thanks,

JonasVautherin, 2022-12-21:
The test showing this message is [here](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/__init__.py#L10-L13):

```python
# Check for compatibility
(major, minor, _) = platform.python_version_tuple()
if not ((int(major) >= 3 and int(minor) >= 6) or (int(major) >= 4)):
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```

Could you try it manually on your platform and see what happens? Maybe there is a bug there but I don't see it.

Nomed73, 2022-12-21:
@JonasVautherin when I try that code alone, I don't get the `[!] MAVSDK-Python is only available on Python >= 3.6` message.

JonasVautherin, 2022-12-21:
Are you sure that you are running the examples from python 3.10? Maybe you have another version (2.x) that is being used instead?

You could try to run in a venv to be sure.

Nomed73, 2022-12-21:
@JonasVautherin In the virtual environment in PyCharm, I did not get that message. I did get the message: 

`ModuleNotFoundError: No module named 'mavsdk'`

If I try using the terminal to run the same file as I did with Pycharm, I still get the message : 

`[!] MAVSDK-Python is only available on Python >= 3.6`

Do you happen to know where `(major, minor, _) = platform.python_version_tuple()` is getting the data from?

JonasVautherin, 2022-12-21:
I would say [here](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/__init__.py#L4) :see_no_evil:. Or do you mean internally in the Python implementation?

Nomed73, 2022-12-21:
I meant internally.  Well that error is not showing anymore. I went through the install all over again, however, now I have the following issue:

```
[Running] /usr/bin/env python3 "/Users/nm/dev/drone_dev/MAVSDK-Python/examples/logfile_download.py"
Traceback (most recent call last):
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/examples/logfile_download.py", line 4, in <module>
    from mavsdk import System
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/mavsdk/__init__.py", line 7, in <module>
    from .system import System
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/mavsdk/system.py", line 8, in <module>
    from . import action
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/mavsdk/action.py", line 5, in <module>
    from . import action_pb2, action_pb2_grpc
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/mavsdk/action_pb2.py", line 16, in <module>
    from . import mavsdk_options_pb2 as mavsdk__options__pb2
  File "/Users/nm/dev/drone_dev/MAVSDK-Python/mavsdk/mavsdk_options_pb2.py", line 21, in <module>
    _ASYNCTYPE = DESCRIPTOR.enum_types_by_name['AsyncType']
AttributeError: 'NoneType' object has no attribute 'enum_types_by_name'

[Done] exited with code=1 in 0.239 seconds
```

I am not really sure what happened. A week ago all was fine. No updates to anything were done and I started to have some issues.  I think I am going to try and start from scratch. Do you recommend that I install python using the download in Python.org or homebrew?  I really appreciate you taking the time to help me.

JonasVautherin, 2022-12-21:
I would use the one from homebrew :+1:

Nomed73, 2022-12-23:
@JonasVautherin I created the venv but continue to get the Attribute Error. I am no longer getting the Pyhon not greater than 3.6 issue.

Nomed73, 2023-01-06:
The file init.py checks for compatibility. However, if you are using python version 3.10 or higher it makes it incompatible.

The line
if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6
is checking float values. So 3.10 and 3.11 are less than 3.60, which is the value the code is checking.

I made the following adjustments to correct for this and this seems to be working.

# Accounting for version 3.10.x and 3.11.x
python_p1 = float(platform.python_version_tuple()[0])
python_p2 = float(platform.python_version_tuple()[1])

if (python_p1 == 2 or (python_p1 == 3 and python_p2 < 6)):
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)

Just in case others were running into this issue.

JonasVautherin, 2023-01-06:
Right, yeah that's what I was suspecting [here](https://github.com/mavlink/MAVSDK-Python/issues/535#issuecomment-1360512723).

Great that you found a solution! Would you mind opening a PR with your changes? That would be helpful, and you would get the credit :innocent:

Nomed73, 2023-01-06:
@JonasVautherin I am not sure how to do that.  :)

JonasVautherin, 2023-01-06:
Oh, that's easy! You need to:

1. Fork the repo (from https://github.com/mavlink/MAVSDK-Python in the upper right corner)
2. Clone your fork on your computer, e.g. `git clone git@github.com:nomed73/mavsdk-python`
3. Ideally create a branch for your fix, e.g. `cd mavsdk-python && git branch fix-python-version-detection && git checkout fix-python-version-detection`
4. Make your changes, commit them, push the branch.
5. The terminal output will show you a link in case you want to open a pull request. Follow it. Alternatively, open your browser to your fork (https://github.com/nomed73/mavsdk-python) and you will see a button to create a pull request with your latest branch.
6. Open the PR, and we will see it!

Let me know how it goes, happy to help :blush:.

Nomed73, 2023-01-06:
It seems that what currently appears in github works also, but that is not what the _init_.py file had when I did the mavsdk install. Do you know why that would happen. 

The version that was installed earlier today had the following line:
```
if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6
```

but the version in the mavsdk folder in github shows the following:
```
# Check for compatibility
(major, minor, _) = platform.python_version_tuple()
if not ((int(major) >= 3 and int(minor) >= 6) or (int(major) >= 4)):
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```
So I shouldn't update it with mine. The one there is cleaner, I am just curious as to why the file that got installed on my venv was different.

JonasVautherin, 2023-01-06:
Which version of mavsdk did you have? Maybe it was outdated :thinking:

julianoes, 2023-03-28:
I'm assuming this was a stale/old init script and this is fixed now.

closed 2023-03-28 by julianoes