mavlink/MAVSDK-Python #539: Offboard , set_position_ned(PositionNEDYaw) Not holding the position.

Murali-IsPagro, 2022-12-26:
Hello ,

I have been testing Offboard Position NED code in MAVSDK-Python, It is moving to the desired position throughout the execution. But , it is not holding up the position, means, when rolling left/right from  Point A to Point B for 3 m, there is yaw of +/- 5 degree and height also not stable it is going up an down by 40 cm. We have used Downfacing lidar and enabled **EKF2_HGT_MODE** to lidar. 

Note: When testing in Loiter mode, it is very stable.
GPS Sat count: 16
HDOP: 0.7
**MPC_XY_VEL_ALL**=0.5

<sub>self.yaw` =0.0
 await self.drone.offboard.set_position_ned(PositionNedYaw(0.0, 3.0, -2.0, self.yaw))
 await asyncio.sleep(6)</sub>

Questions:
1. Any parameter has to be changed in QGC?
2. Anything to be changed in code.?

julianoes, 2022-12-26:
I would generally say this is more of a PX4 issue and less a MAVSDK issue but anyway, for anyone to help you, we need to see a logfile (uploaded to logs.px4.io).

Murali-IsPagro, 2022-12-26:
Sorry, Attaching  the logfile.

[https://review.px4.io/plot_app?log=699084c9-3e19-426e-9297-6a6332110b47](url)

julianoes, 2022-12-27:
This flight looks really clean actually.

For yaw, I wonder if it could be made a bit more aggressive with tuning. Or alternatively, what tends to work is to tilt the 4 motors slightly, so that the tilt of the motor makes any yaw response stronger, e.g. about 2-3 degrees makes a big difference, and if they are off 2 degrees in the wrong direction, it performs quite poorly.

For the altitude control using the lidar, could it be that the ground is not perfectly flat and that's appearing in the air? Or is the tuning just not aggressive enough? I don't know honestly. You could ask in PX4 Discord, maybe someone over there has a good grasp on that.

Murali-IsPagro, 2023-01-14:
Hi, 

We tuned the drone further and tested on a plain surface, now altitude relative to ground change was minimal. It was the problem with uneven surface. 

Thanks

closed 2023-01-14 by Murali-IsPagro