mavlink/MAVSDK-Python #540: Using python 3.10 or 3.11 may cause [!] MAVSDK-Python is only available on Python >= 3.6 issue

Nomed73, 2023-01-06:
The file __init__.py checks for compatibility.  However, if you are using python version 3.10 or higher it makes it incompatible.

The line 
`if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6`
is checking float values. So 3.10 and 3.11 are less than 3.60, which is the value the code is checking. 

I made the following adjustments to correct for this and this seems to be working. 

```
# Accounting for version 3.10.x and 3.11.x
python_p1 = float(platform.python_version_tuple()[0])
python_p2 = float(platform.python_version_tuple()[1])

if (python_p1 == 2 or (python_p1 == 3 and python_p2 < 6)):
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```

Just in case others were running into this issue.

JonasVautherin, 2023-01-06:
Oh, I just answered on the other issue. Same thing here: would you mind opening a PR with your proposed changes? :innocent:

closed 2023-01-06 by Nomed73

JonasVautherin, 2023-01-06:
Seems like it was due to an outdated version of mavsdk-python.

alexbhandari, 2024-03-18:
On ARM based Macs the latest available version from pip is 0.12.0 (see https://github.com/mavlink/MAVSDK/issues/1989 ), which has this problem. So this fix was useful on those platforms where this bug is present and would prevent import