mavlink/MAVSDK-Python #541: pip3 install mavsdk different __init__.py compared to same file in GitHub

Nomed73, 2023-01-06:
I did two test installs yesterday and today. One in PyCharm and one in VS Code. In both cases the mavsdk was installed in the venv. 

When the pip3 install mavsdk is performed, there is the sdk being downloaded from? 

The reason that I ask is because the code on the __init__.py file was different in the install compared to what the file shows in the GitHub repository. 

The lines of code that I am referring to are the following. 
**This is from __init__.py in mavsdk folder in GitHub...**
```
# Check for compatibility
(major, minor, _) = platform.python_version_tuple()
if not ((int(major) >= 3 and int(minor) >= 6) or (int(major) >= 4)):
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```

**This is from the __init__.py in mavsdk folder after installing it in a virtual environment.**
```
# Check for compatibility
if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6:
    print("[!] MAVSDK-Python is only available on Python >= 3.6")
    sys.exit(1)
```

The second one will cause an issue when using python 3.10 or greater since it compares float values. 

This happened when I created a project in PyCharm and one in VS Code with a virtual environment. Is there a place where I can check to see which mavsdk is being installed with pip3? Is there a way that this can be corrected? And lastly, what is the latest version of mavsdk-python, is it 0.12.0? 

Thanks,

JonasVautherin, 2023-01-06:
> When the pip3 install mavsdk is performed, there is the sdk being downloaded from?

It is coming from [PyPi](https://pypi.org/project/mavsdk/#history), the latest version being 1.4.1, which seems [correct](https://github.com/mavlink/MAVSDK-Python/blob/1.4.1/mavsdk/__init__.py#L10-L13).

You can do `pip install --upgrade mavsdk` to update to the latest version.

Nomed73, 2023-01-06:
So the mavsdk-python version should show version 1.4.1 also? When I run ` pip3 list ` it shows mavsdk version 0.12.0. If I do the upgrade suggested, nothing happens. It states that requirements are already met. 

```
(.venv) nm@Noes-MacBook-Pro TelemetryTestVSCode % pip3 list
Package    Version
---------- -------
aiogrpc    1.8
grpcio     1.51.1
mavsdk     0.12.0
pip        22.3.1
protobuf   3.20.3
setuptools 65.5.0
(.venv) nm@Noes-MacBook-Pro TelemetryTestVSCode % pip3 install --upgrade mavsdk
Requirement already satisfied: mavsdk in ./.venv/lib/python3.11/site-packages (0.12.0)
Requirement already satisfied: protobuf>=3.5.1 in ./.venv/lib/python3.11/site-packages (from mavsdk) (3.20.3)
Requirement already satisfied: aiogrpc>=1.5 in ./.venv/lib/python3.11/site-packages (from mavsdk) (1.8)
Requirement already satisfied: grpcio>=1.11.0 in ./.venv/lib/python3.11/site-packages (from mavsdk) (1.51.1)
(.venv) nm@Noes-MacBook-Pro TelemetryTestVSCode % pip3 list
Package    Version
---------- -------
aiogrpc    1.8
grpcio     1.51.1
mavsdk     0.12.0
pip        22.3.1
protobuf   3.20.3
setuptools 65.5.0
```

JonasVautherin, 2023-01-06:
Yeah it should be 1.4.1 :thinking:. You can see the files [here](https://pypi.org/project/mavsdk/#files): 
mavsdk-1.4.1-py3-none-macosx_10_9_x86_64.whl. Do you have an M1/M2 maybe?

Nomed73, 2023-01-07:
Yes. I keep forgetting about the issues with the M1. Well, if people have an M1 and they try to do this, at least there are some solutions for them.

Nomed73, 2023-01-09:
@JonasVautherin and @julianoes what is the process to be able to get the latest mavsdk working on Macs with M1/M2? I would like to assist if I am able to. I just don't know how to get started in contributing.

julianoes, 2023-01-09:
Good question. 

So, to produce an arm-binary we're blocked on GitHub Action to support that: https://github.com/actions/runner-images/issues/2187

Alternatively, we could use an EC2 instance on Amazon based on M1 but we could only spin it up when required, otherwise it would be too expensive. Also, instances on macstadium.com are fairly expensive.

So, in the meantime, we should just ship the Intel binaries which run using Rosetta2, I think.

Now, it seems like that's not happening correctly, and I'm not quite following why. Is it that the wrong thing is pulled from PyPi or do we not upload the correct one? I haven't looked into that in detail.

Nomed73, 2023-01-09:
Can you explain what this part is supposed to do?

> So, in the meantime, we should just ship the Intel binaries which run using Rosetta2, I think.
>
> Now, it seems like that's not happening correctly, and I'm not quite following why. Is it that the [...]

Does this mean that when a user does pip3 install mavsdk that it is supposed to go fetch the latest version (1.4.x) regardless if it is a Mac M1/M2?

closed 2023-01-09 by Nomed73

JonasVautherin, 2023-01-09:
>  Is it that the wrong thing is pulled from PyPi or do we not upload the correct one? I haven't looked into that in detail.

I think (but that's a guess) that pip on the M1/M2 just doesn't find a file with the right name, and gives up. [Those are the files](https://pypi.org/project/mavsdk/#files), where the macOS one is 
`mavsdk-1.4.1-py3-none-macosx_10_9_x86_64.whl`. That's why it falls back to 0.12.0, which [seems to be the last one that had a .whl for architecture "any"](https://pypi.org/project/mavsdk/0.12.0/#files): `https://pypi.org/project/mavsdk/0.12.0/#files`. I don't remember exactly why we removed that, my guess is that it was taking precedence over more specific archs and that was a problem.

If the x86_64 mavsdk_server binary "just runs" on M1/M2, maybe we could just push another file with whatever architecture name goes for M1/M2 (instead of `x86_64`), e.g. `mavsdk-1.4.1-py3-none-macosx_10_9_aarch64.whl` (or whatever the architecture is called on the M1).

@Nomed73: does it work if you download [mavsdk-1.4.1-py3-none-macosx_10_9_x86_64.whl](https://files.pythonhosted.org/packages/92/41/8c4aaa1ec355aa1b85b675e6be080fccd4e7854d54f8ecc48ab68eb1f38d/mavsdk-1.4.1-py3-none-macosx_10_9_x86_64.whl) and try to run `pip install mavsdk-1.4.1-py3-none-macosx_10_9_x86_64.whl`? My hope is that it complains about the architecture (x86_64). If that was the case, you could try to rename x86_64 into the arch of the M1 and try to install it again, and then check if that works.

Does that make sense?