mavlink/MAVSDK-Python #548: Possibility of a landing and a takeoff in the middle of a mission?

Moseskota, 2023-01-24:
Hello @julianoes 

Is it possible to create a mission in which there is a landing and takeoff at a certain waypoint and continuing the mission from there on?
I have been trying to do the same thing by taking **mission.py**  and not **mission_raw.py** example. Landing at a certain waypoint and taking off has been possible but after the takeoff it is not executing next mission_item. 
Can you please provide any insight on how it could be handled or provide an example ?

PS:  I am new to asynchronous programming and drones as well.

julianoes, 2023-01-25:
One problem is that it will auto-disarm when landed, so you would have to disable auto-disarm. Or you'd have to disarm and then wait for some command to arm again. That's not really fully specced and implemented as far as I know. It would require some work on the PX4 side I believe, and also some clarification for the mission protocol.

There was a bit of discussion around this here: https://github.com/mavlink/mavlink/pull/1867

Moseskota, 2023-01-26:
> One problem is that it will auto-disarm when landed, so you would have to disable auto-disarm. Or you'd have to disarm and then wait for some command to arm again. That's not really fully specced and implemented as far as I know. It would require some work on the PX4 side I believe, and also some clarification for the mission protocol.
> 
> There was a bit of discussion around this here: [mavlink/mavlink#1867](https://github.com/mavlink/mavlink/pull/1867)

@julianoes Thank you very much for the prompt reply. Will be waiting for that feature

closed 2023-01-26 by Moseskota