mavlink/MAVSDK-Python #549: want to write an example code of mission_raw

pwl24, 2023-01-30:
Hi I would like to run the drone mission with the local position, I use the mission_row i got the unsupported error from it
i set the param as below, 
mission_items = []
    mission_items.append(MissionItem(0,
                                     1,
                                     16,
                                     1,
                                     1,
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     float('nan'),
                                     25,
                                     25,
                                     -25,
                                     0))
I am running the simulation with jmavsim
or is there any mission_raw example code like mission , That will me helpful for me as well

pwl24, 2023-01-30:
> Hi I would like to run the drone mission with the local position, I use the mission_row i got the unsupported error from it i set the param as below, mission_items = [] [...]

closed 2023-01-30 by pwl24

julianoes, 2023-01-30:
Please create a minimal example for me to reproduce, and put code between ```.