mavlink/MAVSDK-Python #550: Invalid argument error with mission_items in mission_raw.py

Moseskota, 2023-01-30:


julianoes, 2023-01-30:
Ok, the `seq` and `current` fields were wrong, here is what uploads for me:

```
#!/usr/bin/env python

import asyncio
from mavsdk import System
from mavsdk.mission_raw import MissionItem

async def run():
    drone = System(mavsdk_server_address="localhost")
    await drone.connect()

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print_mission_progress_task = asyncio.ensure_future(
        print_mission_progress(drone))

    # included to add redundancy
    running_tasks = [print_mission_progress_task]
    termination_task = asyncio.ensure_future(observe_is_in_air(drone, running_tasks))

    mission_items = []
    mission_items.append(MissionItem(0,6, 22, 1, 1, 0, 10,  0, float('nan'),  169385325,  817945425, 10.0, 0))
    mission_items.append(MissionItem(1,6, 16, 0, 1, 0, 10,  0, float('nan'),  169388925,  817942725, 10.0, 0))
    mission_items.append(MissionItem(2,2, 20, 0, 0, 0,  0,  0,   0,           169385025,  817940325, 10.0, 0))

    await drone.mission_raw.upload_mission(mission_items)

    print("mission uploaded")
    await asyncio.sleep(2)

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    await drone.action.arm()
    print("arming the done")

    await drone.mission_raw.start_mission()
    print("started the mission")

    await termination_task

async def print_mission_progress(drone):
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: "
              f"{mission_progress.current}/"
              f"{mission_progress.total}")

async def observe_is_in_air(drone, running_tasks):
    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

    if was_in_air and not is_in_air:
        for task in running_tasks:
            task.cancel()
            try:
                await task
            except asyncio.CancelledError:
                pass
        await asyncio.get_event_loop().shutdown_asyncgens()

        return

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())

```

Moseskota, 2023-01-31:
> Ok, the `seq` and `current` fields were wrong, here is what uploads for me:
>
> ```
> #!/usr/bin/env python
>
> import asyncio
> from mavsdk import System
> from mavsdk.mission_raw import MissionItem
>
>
>
>
> async [...]

> Ok, the `seq` and `current` fields were wrong, here is what uploads for me:
>
> ```
> #!/usr/bin/env python
>
> import asyncio
> from mavsdk import System
> from mavsdk.mission_raw import MissionItem
>
>
>
>
> async [...]

Thank you very much for the prompt reply. But i have a question. For ardupilot , i saw that the seq starts with 0 whereas for px4 it starts with 1.

Moseskota, 2023-01-31:
@julianoes 
Thank you very much for your prompt reply.

closed 2023-01-31 by Moseskota