mavlink/MAVSDK-Python #554: Injecting motor failure produces wrong output?

m-sienk, 2023-02-04:
Docker container: px4io/px4-dev-ros2-foxy
PX4 version=latest
mavsdk version=1.4.3

Injecting the following:

`await system.failure.inject(FailureUnit.SYSTEM_MOTOR, FailureType.OFF, instance=0)`
or
`failure motor off -i 0`

telemetry.health produces is_armable: False:
`Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: True, is_armable: False]`

telemetry.health_all_ok remains True:
`'health_all_ok': True`

Should 'health_all_ok' not produce False in this case?

Also, is this currently the only way to detect a prop failure? In the future will there be a way to detect individual prop's status so I am able to detect which prop has failed? 
I am thinking of airframes that can sustain a single prop loss but then need to engage in some custom emergency landing behavior once a failure is detected.

julianoes, 2023-02-07:
Not all failures are supported on the PX4 side. Feel free to implement what you need. To debug which motor has failed exactly you will have to use other information such as the [ESC Status message](https://mavlink.io/en/messages/common.html#ESC_STATUS). 

Now regarding `health_all_ok`, that's probably suboptimal. MAVSDK doesn't really know all the possible "health" metrics, so we should probably phase out and remove that meta flag and use `is_armable` as the overall flag to signal something is good to go, since this is produced by the autopilot and signals whether it is ok with respect to everything that is configured/required.

m-sienk, 2023-02-09:
Thanks for your response.

So if I wanted more information regarding the status of the system:
```
is_esc_1_ok
is_esc_2_ok
(... how ever many ESC's...)
```

I also want to track individual peripheral's - if I have 2 GPS units connected (for redundancy) and I want to know:
```
is_gps_1_ok
is_gps_2_ok
```

Or:
```
is_distance_sensor_1_ok
is_distance_sensor_2_ok
(... how ever many distance sensors...)
```

Are there any plans to include such functionality in px4/ mavsdk in the future?

julianoes, 2023-02-09:
I suggest to add the features yourself and make pull requests. It's very far down on my list I'm afraid.

closed 2023-02-19 by m-sienk