mavlink/MAVSDK-Python #555: MAVSDK Initialization and AirSim settings for using SimpleFlight

lmr2693, 2023-02-07:
Hello,

I'm quite green in my coding skills and CS knowledge. Any assistance is appreciated.

Issue: The mavsdk code is not connecting to the AirSim drone. I assume its a combo of AirSim json settings and the MAVSDK initialization code and perhaps the mavsdk_server.

Background: I'm on a drone project tasked with autonomously flying UAVs. I've been tasked with setting up a simulator to test the Team's MAVSDK-python code. We've chosen Unreal Engine with AirSim (on Windows) and successfully installed AirSim and run its example PythonClient code. Note that we are not using a controller, so I assume AirSim is defaulting to its built-in controller "SimpleFlight". Will that be a problem with mavsdk?

[Text file with the settings and code pasted below: [Settings_Code.txt] ](https://github.com/mavlink/MAVSDK-Python/files/10680072/Settings_Code.txt)

My startup steps are as follows:

1) For manually setting the mavsdk_server to 50051, I downloaded the mavsdk_server.exe file and saved it to the mavsdk bin folder. I manually ran the file as follows: 
[image]

I exited the command prompt, and then double clicked the actual file. Now the file runs on a different port. Will this be a problem?
[image]

2)  How should we set the Airsim settings.json file? Is it a problem that we are using AirSim's flight controller and not a PX4? We were hoping to not have to use anything that would require running linux.

Current AirSim settings:
{
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SimMode": "Multirotor",
  "LocalHostIp": "127.0.0.1",
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "LocalHostIp": "127.0.0.1",
      "ApiServerPort": 41451,
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "EnableTrace": false,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      }
    }
  }
}

3) To test AirSim, I manually ran mavsdk again (step 1 above), then left the command window open, and ran the MAVSDK code below.
(Note: I was able to run AirSim with their PythonClient code.)

import mavsdk
import asyncio
from mavsdk import System

async def run():
    drone = mavsdk.System(mavsdk_server_address='localhost', port=50051)
    print("Using mavsdk server port 50051")

    print("Waiting for drone to connect...")
    await drone.connect(system_address = "udp://:14540")

    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered!")
            break
    
    print("-- Arming")
    await drone.action.arm()
    await asyncio.sleep(5)
    print("Set init poit to 0 0 0")
    await drone.offboard.set_position_ned(PositionNedYaw(0.60, -4.57, 0.0, 180))

    print("Local coordinate system set")

    print("Start offboard mode")
    await drone.offboard.start()

    print("Taking off now")
    print("--First motion")
    await drone.offboard.set_position_ned(PositionNedYaw(0.60, -4.57, -1.5, 180))
    await asyncio.sleep(20)
    
    await drone.offboard.set_position_ned(PositionNedYaw(0.60, -4.57, -0.25, 180))
    await asyncio.sleep(5)
    
    print("--landing now")
    await drone.action.land()
    
    await asyncio.sleep(7)

    print("disarming the drone now")
    await drone.action.disarm()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

The outputs are:
[image]

JonasVautherin, 2023-02-08:
1. You are running an old MAVSDK (0.35, we are at 1.4.x)
2. By default MAVSDK listens on `udp://:14540`. I see a [QgcPort](https://github.com/Microsoft/AirSim/blob/main/docs/settings.md#additional-px4-settings) option in AirSim, could you try setting it to 14540? Alternatively you could run mavsdk_server with arg `mavsdk_server -p 50051 udp://:14550` to use the same port as QGC.

Assuming that AirSim runs on the same computer as mavsdk_server, once you change the port, mavsdk_server should detect a system.

lmr2693, 2023-02-08:
1) Thanks for noticing. I'd downloaded the mavsdk_server_win32.exe file from an older issue post. Didn't notice how old the post was. Everything's running on v.1.4 now.

2) QgcPort did not work. :/

Going to install WSL2, PX4, and QGroundControl. If AirSim still an issue, may just scrap airsim and give gazebo a try.

I'll leave this thread open for a couple days, for if I have a question on PX4. 

Thanks!

JonasVautherin, 2023-02-08:
Did you manage to get AirSim to work with QGC, by the way? I mean AirSim should be sending to `127.0.0.1:14550` by default, so QGC should pick it up.