mavlink/MAVSDK-Python #556: Unable to connect Pixhawk to Raspberry Pi using MAVSDK

aadithyasujit, 2023-02-10:
Hi @julianoes, 
I am trying to connect my Pixhawk 4 (running PX4) to Raspberry Pi 3/4 using MAVSDK but end up with the below error. I have installed MAVSDK using the pip3 command following instructions from [here](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/index.html#install-using-pip-from-pypi). 
[image]
I am using an FTDI adapter and connecting the pixhawk to Pi using usb & use the below command to connect to the Pixhawk
[image]

In an attempt to troubleshoot, I am able to receive data on the port when i checked screen using screen /dev/ttyUSB0 and when I try the connecting using using Mavproxy, it works fine. But I have not been able to establish a connection MAVSDK. 
What else would I have to check in order to get this working with MAVSDK?

julianoes, 2023-02-10:
You might need the latest bugfix: https://github.com/mavlink/MAVSDK/pull/1968
but it's not released in Python yet.

Let me do a Python release and then you can try again.

julianoes, 2023-02-10:
Can you try with https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.4. It should be on PyPi shortly.

aadithyasujit, 2023-02-10:
> Can you try with https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.4. It should be on PyPi shortly.

I did try this using `pip3 install mavsdk --upgrade`. 
```
Collecting mavsdk
  Downloading mavsdk-1.4.4-py3-none-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.6 MB)
     |████████████████████████████████| 11.6 MB 1.6 MB/s 
Requirement already satisfied, skipping upgrade: grpcio>=1.50.0 in /home/linux-sim/.local/lib/python3.8/site-packages (from mavsdk) (1.51.1)
Requirement already satisfied, skipping upgrade: aiogrpc>=1.8 in /home/linux-sim/.local/lib/python3.8/site-packages (from mavsdk) (1.8)
Requirement already satisfied, skipping upgrade: protobuf<=3.20.1 in /home/linux-sim/.local/lib/python3.8/site-packages (from mavsdk) (3.18.0)
Installing collected packages: mavsdk
  Attempting uninstall: mavsdk
    Found existing installation: mavsdk 1.4.3
    Uninstalling mavsdk-1.4.3:
      Successfully uninstalled mavsdk-1.4.3
Successfully installed mavsdk-1.4.4
```

But the error still remains the same.  

```
maggy@raspberrypi:~/examples $ python3 ./firmware_version.py 
Waiting for drone to connect...
-- Connected to drone!
Traceback (most recent call last):
  File "/home/maggy/examples/./firmware_version.py", line 23, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/maggy/examples/./firmware_version.py", line 17, in run
    info = await drone.info.get_version()
  File "/home/maggy/.local/lib/python3.9/site-packages/mavsdk/info.py", line 746, in get_version
    raise InfoError(result, "get_version()")
mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET: 'Information Not Received Yet'; origin: get_version(); params: ()
```
I did also try manually starting the server based on this post, but no luck

[image]

julianoes, 2023-02-12:
Can you get the firmware version after waiting a few seconds in the script?

aadithyasujit, 2023-03-09:
Hi @julianoes, 

I can confirm that I am running mavsdk-1.4.4. 
I recently tried the telemetry example and it worked all worked fine ! I think the only problem is that the firmware.py returns that error. I can live with it for now. 

Thanks anyways !

arpitpara, 2023-03-22:
> Can you get the firmware version after waiting a few seconds in the script?

No
I waited for 5-10s in the script.
It still say the same.
Although I'm trying with px4 SITL.
But the problem remains same.

julianoes, 2023-03-22:
@arpitpara can you paste the output you see with the mavsdk_server output separate?

arpitpara, 2023-03-22:
> @arpitpara can you paste the output you see with the mavsdk_server output separate?
[image]

This is modified script with sleep time 5 sec to wait for params.

julianoes, 2023-03-22:
It's probably not connected.

You have to debug the connection with the mavsdk_server running separately, as described here:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

arpitpara, 2023-03-23:
No,I'm pretty sure it is connected. Cuz if I run takeoff and land.py it runs perfectly as I can see drone taking off and landing in SITL PX4 Gazebo environment.
[image]

arpitpara, 2023-03-23:
Yeah Sure!
Please let me know of any help. I've also recently started development on the same and wants to learn from you!

julianoes, 2023-03-23:
Ok, I can reproduce the version problem.

closed 2023-03-23 by julianoes

arpitpara, 2023-03-24:
> This should fix it: [mavlink/MAVSDK#2006](https://github.com/mavlink/MAVSDK/pull/2006)
> 
> For now, just ask for the version twice, so ignore the first exception and try again, and that should work.

yes, it's working
[image]
!

julianoes, 2023-03-25:
Yep, just one remark: in Python async, use `await asyncio.sleep(2)` instead of `time.sleep(2)`.

PR #569 (2023-03-28): Bump mavsdk_server version to v1.4.13
This mainly fixes getting info.version on the first try.

Closes #556.