mavlink/MAVSDK-Python #559: Stable Release for PX4 v1.11.3-stable

a-altug, 2023-03-06:
Is there a specific stable version for PX4 v1.11.3?

JonasVautherin, 2023-03-06:
No, but it should work with v1.11.3. Do you have a specific problem?

a-altug, 2023-03-06:
Not actually. But after "multiple ekf" I lost the strings :) I flashed pixhawk with v1.11.3.

JonasVautherin, 2023-03-06:
I'm a bit confused. You are asking here in MAVSDK-Python about the PX4 version you flash on your Pixhawk?

So is this a question about MAVSDK-Python and PX4, or is it solely about PX4 and that's the wrong place to ask? :see_no_evil:

julianoes, 2023-03-06:
Fair question. Generally MAVSDK should be backwards compatible as it's based on the MAVLink API. In this case, I would recommend PX4 v1.13.x (or main/v1.14) and latest v1.x.y MAVSDK-Python.

closed 2023-03-06 by julianoes