mavlink/MAVSDK-Python #562: Invalid argument error with 'import_qgroundcontrol_mission' in mission_raw.py

kennethakiti1, 2023-03-10:
Hi, currently im running MAVSDK-python on Jetson nano. I'm able to get it working for the basic examples, but for the mission_raw.py example, it keeps giving an error.

using the same code as in the example here [mission_import.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission_import.py). works on my windows computer without any error, but when i try to run it on my jetson nano, i get the error below

[image]

it says 'mavsdk.mission_raw.MissionRawError: FAILED_TO_OPEN_QGC_PLAN: 'Failed To Open Qgc Plan'; origin: import_qgroundcontrol_mission(); params: ('mission.plan',)'

julianoes, 2023-03-12:
Hi, it looks like it doesn't find the file. Is the file `mission.plan` in the directory in which you're running the script? So this is not about whether the file is in the same directory but whether the file is in the directory you're in when you run the script.

kennethakiti1, 2023-03-12:
> Hi, it looks like it doesn't find the file. Is the file `mission.plan` in the directory in which you're running the script? So this is not about whether the file is in the same directory but whether the file is in the directory you're in when you run the script.

Yes please, I have the mission file in the same directory as the script. 

I have run this code before on my raspberry pi so i don't know exactly what's happening (for context, this is on a jetson nano). The only difference between that now I'm using the latest version of mavsdk_server.

julianoes, 2023-03-12:
How are you running the mavsdk_server on the Jetson?

I have an idea: use the absolute path to reference the `mission.plan`, e.g. `/home/pi/mission.plan`. The problem could be that the directory that the server runs in is different from the Python script. The file gets loaded in the server, so it's actually that path that counts.

kennethakiti1, 2023-03-13:
> How are you running the mavsdk_server on the Jetson?
>
> I have an idea: use the absolute path to reference the `mission.plan`, e.g. `/home/pi/mission.plan`. The problem could be that the directory [...]

OMG thanks that works. 
quite strange given that the relative path works on my windows environment and on my raspberry pi. cant believe i didnt try to use the absolute path. once again thanks. 

for anyone wondering try this 
`mission_import_data = await drone.mission_raw.import_qgroundcontrol_mission("/home/nano/mission_upload_test/mission.plan")`

closed 2023-03-13 by kennethakiti1