mavlink/MAVSDK-Python #563: Unable get position body and imu information through Raspberry Pi

Hyc1023, 2023-03-18:
Hi @julianoes ,
I am working on a algorithm that will need the body position from odometry under telemetry. When I test my code with Gazebo, function odometry works just fine. But when I try to test my code on board with Raspberry Pi connected with the PixHawk, I couldn't get any information of the body position and my code just stuck at the function waiting to get the body position. The drone system is properly connected because I can run takeoff mission through my code. 

I'm wondering that does odometry only works under virtual machine like Gazebo, or do I miss checking any needed information to get the body position.

Since I can't get the body position from odometry, I tried to use imu's acceleration to calculate displacement instead of body position and the same problem occurs.  

This is the testing code I rewrite from the example code telemetry.py, and I put it on Raspberry Pi to test but I can't get the information I wanted.
[image]

julianoes, 2023-03-19:
Odometry is only sent out by a vehicle that actually estimates odometry. I think that's only the case if you have some sort of external vision or similar hooked up to PX4, so that the EKF2 can estimate odometry.

What information do you actually require? Global position (latitude/longitude from GPS) or local position (meters from estimator initialization)?

Hyc1023, 2023-03-20:
First of all, thank you so much for your immediate reply.
Do you mean if I need the information (local position) from odometry, I'll have to connect something that estimates odometry with my Pixhawk? I thought at the position where I takeoff, odometry will initial the position coordinate (x, y) as (0, 0), and calculate displacement with imu or something else in PixHawk.

At my current situation that I simulated on Gazebo, I need odometry's position_body (x_m, y_m, z_m) to initialize the local position at where I takeoff. Furthermore, every time I move my drone to a new position, I use position_body to get the new local position and calculate displacement with the previous local position. I use local position to calculate the displacement instaed of velocity because there will be interference from external factors no matter on Gazebo or in reality.  

In addition to the previously mentioned issue that I couldn't get the local position, I found out if I takeoff at coordinate point (8, 0) in Gazebo, the initial local position I get will also be (8, 0). Does it mean if I use it on board I'll have to give a initial coordinate point since there's no virtual coordinate in reality? Or does it mean if I use class odometry properly, the initial local position will be (0, 0)?

julianoes, 2023-03-20:
Please take a step back and describe what you're actually want to do. I think the whole odometry topic is just a bit of a confusion/distraction here. It's something that is available in SITL but not something that you can use easily overall.

Are you flying indoor or outdoors (with GPS)?

Hyc1023, 2023-03-21:
Currently I'm flying outdoors with GPS,  and I'm trying to record the drone's local position every sampling time including the local position where it takeoff.

julianoes, 2023-03-21:
Aha, you should use: [position_velocity_ned](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Telemetry.position_velocity_ned) instead.

Hyc1023, 2023-03-22:
I tried it and it works, thanks a lot. 
By the way, I found out that it seems to be calculating the local position with the GPS signal. Is there any function that does the same thing under GPS-denied situation?

closed 2023-03-22 by julianoes