mavlink/MAVSDK-Python #570: Flight mode does not change after entering "return to launch" mode

kennethakiti1, 2023-03-28:
I'm currently using the jetson nano with the mavsdk_server. I'm able to get the drone to go on missions etc, but whenever I initiate a return to launch command, the flight mode gets stuck in this mode and unable to take other new commands such as "takeoff", "kill", "begin_mission" etc unless I use the RC controller to switch the mode before it can begin another action. 

is there a way to set the flight mode, let's say from "return to launch" to stabilized mode or another mode?

julianoes, 2023-03-29:
Are you trying to change the flight mode when it's still flying or once landed and disarmed?

You can set flight modes like this:
- mission->start() for mission mode
- mission->pause() for hold mode
- manual_control->start_position_control() for position control
- manual_control->start_altitude_control() for altitude control

kennethakiti1, 2023-03-29:
I am trying to change the flight mode once it's landed and disarmed. 

 Actually, I don't have much interest in changing the flight mode, but since after landing and disarming it gets stuck in the _"RETURN_TO_LAUNCH"_ mode, I need to find a way of changing the mode withing needing the remote controller.

julianoes, 2023-03-29:
It should still let you change the flight mode once landed. Do you see any errors when you try to change it? E.g. status text messages or messages in QGC?