mavlink/MAVSDK-Python #571: Action vs ActionServer?

m-sienk, 2023-03-30:
Hi,

What is the difference between the 'standard' and 'server' implementation of a plugin; for instance the 'Action' plugin and the 'ActionServer' plugin.

My guess is that the answer here lies in the naming convention in which the intention may be to use the 'Action' plugin when writing an application that runs on an onboard computer and where the 'ActionServer' plugin is used when writing a GCS type of application? 

Thanks

JonasVautherin, 2023-03-30:
`Action` is the "client". For instance, for `action.takeoff()`, the client is the GCS and the "server" is the autopilot. You would use `action_server` to write an autopilot (or anything that could receive the "takeoff" command from a client).

So I think that it is the opposite of your intuition :see_no_evil:.

Does that help?

m-sienk, 2023-03-31:
Thanks for your response, sorry I'm a bit confused.

ActionServer is missing many commands that are available to the Action plugin i.e: do_orbit, goto_location, ... etc

Let me explain what I have been experimenting with and maybe you can tell me how it should fit together.

I  built an aircraft that runs on the LTE network using MQTT as a messaging protocol. I have my own messaging format in JSON in which I pack some extra data into the message - the message also contains the command that I want the aircraft to execute like arm, takeoff etc.

I’m using a raspberry pi which is connected via a serial connection to a Pixhawk. 
On the raspberry pi I am running a ‘drone-client’ a python program which contains an mqtt-client and mavsdk-python. The client receives the JSON message de-serializes it and then uses mavsdk-python to communicate the command to the Pixhawk.

Does ActionServer fit into this scenario?
[image]

Thanks

JonasVautherin, 2023-03-31:
> Does ActionServer fit into this scenario?

I don't see why you would need it :thinking:. Isn't `drone-client -> mavsdk.action.arm() -> PX4` enough?

m-sienk, 2023-03-31:
Yes, I thought so.

I'm just trying to work out a scenario in which the server implementation is useful.  

Is there any sample code using action server ?