mavlink/MAVSDK-Python #572: camere TIMEOUT error when running the example camera.py

Radyii, 2023-04-07:
Hi i am when trying to set camera mode or take a photo the camera throw the error TIMEOUT, even using the camera.py example. In beginning thought it was problem with PX4 gazebo simulation so I recloned the repo, but that didn't fix the issue.

To make sure  its something to do with mavsdk I tried to take a photos using QGroundControl and that was successful and the photo event appeared on the px4 console, so it seem it's an issue with mavsdk.

Also sometime (maybe once in 5 run) the camera.py throw this error .
```
`Waiting` for drone to connect...
-- Connected to drone!
Setting mode to 'PHOTO'
Camera mode: UNKNOWN
Setting mode failed with error code: TIMEOUT
Taking a photo
Couldn't take photo: TIMEOUT
exception calling callback for <Future at 0x7f84ca68b400 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.8/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 881, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
exception calling callback for <Future at 0x7f84c8564100 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.8/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 881, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
```

[image]

JonasVautherin, 2023-04-07:
> I tried to take a photos using QGroundControl and that was successful

Hmm I'm a bit confused. SITL does not have a camera module, does it?

Radyii, 2023-04-07:
Not sure about other simulator  but Gazebo has a simulated Mavlink camera plugin mentioned [here](https://docs.px4.io/v1.12/en/simulation/gazebo.html#simulated-survey-camera) and implemented [here](https://github.com/PX4/PX4-SITL_gazebo-classic/blob/main/src/gazebo_camera_manager_plugin.cpp).
using QGroundControl result in an photos capture in the px4 shell so I don't think the issue us from the  SITL simulation. 
[image]

JonasVautherin, 2023-04-07:
Interesting, I did not know that :thinking:. Do you have to enable this camera plugin, or is it always there by default?

Radyii, 2023-04-08:
> Do you have to enable this camera plugin, or is it always there by default?

It's enable by defaults when using a model/frames that support it (typhoon h480, plane_cam...) , it detect that Gstreamer is installed and add the plugin automatically.

[image]

Also  since I am sure now that it was working before i tried using different models (typhoon h480, plane_cam...), when that didn't work, I setup the whole environment on a fresh Ubuntu install  on a remote Vps first, then on a local machine but it's still the same issue. In fact i am getting this error every time i run camera.py now.

```
`Waiting for drone to connect...
-- Connected to drone!
Setting mode to 'PHOTO'
Camera mode: UNKNOWN
Setting mode failed with error code: TIMEOUT
Taking a photo
Couldn't take photo: TIMEOUT
exception calling callback for <Future at 0x7fd44c1c17f0 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.8/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 881, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
exception calling callback for <Future at 0x7fd44811ef70 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.8/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 475, in __next__
    return self._next()
  File "/usr/local/lib/python3.8/dist-packages/grpc/_channel.py", line 881, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed`
```

Lastly this Gazebo's warning keep popping up so it might be related, but still QGroundControl succeed at sending the capture photo Mavlink message with no issue.

`[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.`

JonasVautherin, 2023-04-08:
Ok, I did not know about that "survey camera" plugin. Here it seems like MAVSDK sends the message to take a photo, but does not receive the expected answer.

MAVSDK sends [MAV_CMD_IMAGE_START_CAPTURE](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/camera/camera_impl.cpp#L314), do you know what QGC sends?

Could you somehow debug on the SITL side to see if the MAVSDK message (MAV_CMD_IMAGE_START_CAPTURE) reaches SITL? Does the survey camera plugin support this message?

Radyii, 2023-04-09:
So I managed to fix the issue by uninstalling modemanager and adding the $user to the Dialout group (recommended by QGroundControl install page, but since i had a local QgroundControl AppImage I forgot about it ).

> Could you somehow debug on the SITL side to see if the MAVSDK message (MAV_CMD_IMAGE_START_CAPTURE) reaches SITL? Does the survey camera plugin support this message?

 Thanks Jonas , this what led me to figure up the cause.

Still this doesn't explain why QGroundControl was succeeding in sending the command and taking the photo?

 For future reference  I will attach the Mavlink's messages sent by mavsdk when running camera.py (captured using wireshark) [mavsdk_Camera_Mavlink_messages.csv](https://github.com/mavlink/MAVSDK-Python/files/11184764/mavsdk_Camera_Mavlink_messages.csv) and QGroundControl camera capture mavlink's messages[QGroundControl_Camera_Mavlink_messages.csv](https://github.com/mavlink/MAVSDK-Python/files/11184765/QGroundControl_Camera_Mavlink_messages.csv).

But since my problem is solved i will close the issue.

closed 2023-04-09 by Radyii

JonasVautherin, 2023-04-09:
> Still this doesn't explain why QGroundControl was succeeding in sending the command and taking the photo?

Did you check if they use the same message? Would be interesting to know :+1:

Radyii, 2023-04-10:
> Did you check if they use the same message? Would be interesting to know 

Yeah when sending mavlink message for setting "photo" mode and capture they are sending the same messages (`COMMAND_LONG: MAV_CMD_SET_CAMERA_MODE`, `COMMAND_LONG: MAV_CMD_IMAGE_START_CAPTURE`).

The real cause of the issue is not the content of the Mavlink's messages, but the port used by Mavsdk and QGroundControl. Mavsdk use the offboard port 14540 to send messages to port 14580 which reply with `COMMAND_ACK: MAV_CMD_SET_CAMERA_MODE `and `COMMAND_ACK: MAV_CMD_IMAGE_START_CAPTURE`.
[image]

While  QGroundControl use the GCS port 14550 to send the Mavlink messages to port 18570 which are , according to [this](https://docs.px4.io/main/en/simulation/#camera-simulation), forwarded by PX4 to the simulated MAVlink camera port 15430.
[image]

So from my current understanding it's a PX4 issue of forwarding GCS(port: 14550) Mavlink camera messages to the simulated camera, and not forwarding the messages coming from the offboard api (port: 14540).
Using port 14550 to connect to the drone result in Camera.py example running perfectly. 

A solution is forwarding all MAVlink messages(   that's why I thought "uninstalling modemanager and adding the $user to the Dialout group", were what fixed the issue, while in fact QGroundControl was forwarding the traffic ).