mavlink/MAVSDK-Python #582: MAVSDK-Python apython stuck and server waiting

AdrianVitoria, 2023-04-20:
Wanted to connect to a Mission Planner Simulation to do the basics of arm, take-off and land.
I have little experience with Python so I followed a guide: [Link](https://mavsdk.mavlink.io/main/en/python/quickstart.html)
Using the "apython" thing, I tried to connect many times, to no avail.
And now it's stuck and have to close the terminal every time I try it.

How I'm opening the server: mavsdk_server_win32 -p 50051
Result:
["Time"|Info ] MAVSDK version: v1.4.12
["Time"|Info ] Waiting to discover system on udp://:14540...

Waits forever.

What I do in apython:
>>> from mavsdk import System
>>> drone = System(mavsdk_server_address="localhost, port=50051")
>>> await drone.connect(system_address="udp://:14540")

Stays stuck right there.

What am I doing wrong?
Please help.

AdrianVitoria, 2023-04-20:
Don't know much about Mission Planner but I'm asuming it's this thing that says TCP and 115200.
[image]

I'll try to fiddle with the options and see if I can change those to UDP and 14540.

JonasVautherin, 2023-04-20:
Doesn't the Ardupilot SITL listen on TCP 5790 by default, or something like this?

You could try something like this to connect MAVSDK to it:

```
mavsdk_server_win32 -p 50051 tcp://127.0.0.1:5790
```

AdrianVitoria, 2023-04-20:
5790 gave a socket error (?)
5760 is waiting.
Changed the connect to: await drone.connect(system_address="tcp://127.0.0.1:5760").
Stuck again.