mavlink/MAVSDK-Python #583: Takeoff without GPS and control device

cyb3rL1ght, 2023-04-20:
Good afternoon dear developers!!!
**_Remark_**: I live in a region where GPS catches poorly due to bad weather! it's connected, but I wouldn't want to use it!
**_Question_**: How can I take off without using GPS and radio control? I want to use only mavsdk_server + pycharm on my laptop
Go....I have:
Pixhawk 6x with GPS M8 connected to it, Eth cable for connection to a companion computer, esc, motors (schematic diagram is shown below)
[image]

The connection is established using the MAVLINK-ROUTER (the connection works fine) **AND THERE ARE NO OTHER COMMUNICATION HARDWARE!!!**(no radio control)

[image]

After turning on the quadcopter, I get an error message (screenshot below)
[image]

After pressing the ARM button, I see another message:
“Arming denied! manual control lost”
[image]

And I can’t do anything else! I tried to change the parameter COM_RC_IN_MODE = 4, COM_ARM_WO_GPS, but that didn’t help either!
[image]
[image]

julianoes, 2023-04-21:
This is more of a PX4 question than a MAVSDK one.

However, given you've a lot of detail which is helpful, I'll try to give some hints.

I think the problem is that you're trying to arm in Manual mode, which is - as the name suggests - a manual name, so it needs "manual" input. If you have neither RC nor a joystick in QGC configured, then you don't have any manual input and therefore it just doesn't work. How would it know where/how to fly?

julianoes, 2025-05-01:
No more follow up, closing.

closed 2025-05-01 by julianoes